cyberbotics/webots #1805: reset environment

culurciello, 2020-06-18:
Hi, I am trying to reset the environment with a reset() function, and am setting some random position of a target in the same function with: target_pos.setSFVec3f(new_pos). The target position changes if I do not run the function: simulationReset(), but if I do, it takes only the original position from the wet file... is there a way to reset the simulation or robot position and at the same time set some random positions for a target? Thanks a lot!
E

omichel, 2020-06-18:
You should probably perform one simulation [step](https://cyberbotics.com/doc/reference/robot#wb_robot_step) after the simulation reset before setting the new position for your objects.

culurciello, 2020-06-18:
thanks a lot, that was my problem!

closed 2020-06-18 by culurciello