cyberbotics/webots #1807: Move ros_python controller in webots_ros as extern controller

DavidMansolino, 2020-06-19:
**Describe the Bug**
This controller https://github.com/cyberbotics/webots/tree/master/projects/languages/ros/controllers/ros_python

Should be moved to: https://github.com/cyberbotics/webots_ros
This will save us from maintaining and releasing the ros python libraries.
Furthermore, this is causing issues when creating ROS2 releases as the automatic test of the build farm are detecting that we are embedding ROS1 stuff and doesn't like it.

Simon-Steinmann, 2020-06-28:
@DavidMansolino 
As an input from my side, having dived deep into webots and ROS:
It is much easier to simply use the extern controller functionality of webots. Having a nice little working ROS controller inside webots is nice, but anyone actually wanting to use ROS inside of webots, will run into many roadblocks. Editing runtime.ini files, changing system variables and perhaps launching webots through the terminal in a custom environment can be confusing. 

I think it would be simpler and easier to understand, if the official ROS workflow and setup is explained as using an extern controller. It is very intuitive and would probably lead to less questions and confusions.

DavidMansolino, 2020-06-29:
@Simon-Steinmann 
Thank you for your feedback, I completely agree with you, having everything on the Webots side was a design issue on our side, we are by the way improving this for our ROS2 interface (https://github.com/cyberbotics/webots_ros2) where everything is implemented as an extern controller.

DavidMansolino, 2020-07-09:
This is now fixed in #1876.

PR #1876 (2020-07-09): Move ros_python controller in webots_ros as extern controller
**Description**

Fix #1807.
closed 2020-07-09 by DavidMansolino