cyberbotics/webots #1819: Debugging symbols are removed on compile

ljden, 2020-06-23:
**Describe the Bug**
Debugging symbols are stripped at linking when following the Webots guide for [Using the GNU Debugger with a Controller](https://cyberbotics.com/doc/guide/debugging-c-cpp-controllers?tab-os=linux&tab-language=c++#using-the-gnu-debugger-with-a-controller)

**Steps to Reproduce**
1. Create dummy controller
2. Open Makefile in controller directory and add line `CFLAGS = -g`
3. Save and compile with `make`
4. run `gdb dummy_controller`
5. gdb outputs:

> GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./dummy_controller...(no debugging symbols found)...done.

**Expected behavior**
Debugging symbols should not be stripped so they can be loaded by gdb

**System**
 - Operating System: Ubuntu 18.04
 - Graphics Card: N/A

**Additional context**
Compiling with `make debug` does not strip the debugging symbols

The offending lines are related to the `release` goal: https://github.com/cyberbotics/webots/blob/03742b638505d796a0e87a38c3a1d9d94ad303a6/src/webots/Makefile#L203-L208

If this is still the desired behaviour, then the documentation (linked above) should be updated to recommend compiling with `make debug` and the suggestion of adding `CFLAGS = -g` to the Makefile should be removed

omichel, 2020-06-23:
This is the desired behavior.
You are right, the documentation is not up-to-date to this.
Feel free to propose some fix for the documentation.

ljden, 2020-06-23:
@omichel Is there a way to choose the make target with the "build" button in the Webots app? I can compile with `make debug` from the command line, but I can't see anything obvious in the app to do the same

omichel, 2020-06-23:
No, you are right, this is only possible from the command line.

DavidMansolino, 2020-06-23:
The lines your are referring in `webots/src/webots/Makefile` are for compiling Webots executable and not for compiling controllers.

ljden, 2020-06-23:
Ah you're right! They are (nearly) the same in the [`Makefile.include` file](https://github.com/cyberbotics/webots/blob/03742b638505d796a0e87a38c3a1d9d94ad303a6/resources/Makefile.include#L268-L275) hence my confusion

ljden, 2020-06-23:
@omichel I've edited the documentation in a fork and am making a PR, should that go into master?

omichel, 2020-06-23:
Yes.

closed 2020-06-23 by omichel