cyberbotics/webots #1821: Crash when opening TurtleBot3Burger in scene tree on develop branch

omichel, 2020-06-23:
On Windows, on the `develop` branch, if you open the turtlebot3 (projects/robots/robotis/turtlebot/worlds/turtlebot3_burger.wbt) and double-click on the TurtleBot3Burger node in the scene tree, Webots crashes with the following stack:
```bash
#0  0x000000006fd4002a in libstdc++-6!.dynamic_cast ()
   from D:\msys64\mingw64\bin\libstdc++-6.dll
#1  0x000000000064f257 in WbFieldModel::isMultiple (
    this=0x12bd0560 <QArrayData::shared_null>) at vrml/WbFieldModel.cpp:257
#2  0x00000000004ade00 in WbField::isMultiple (this=0x3080bc30)
    at vrml/WbField.cpp:290
#3  0x0000000000588beb in WbSceneTree::handleDoubleClickOrEnterPress (this=
    0x19374730) at scene_tree/WbSceneTree.cpp:1417
#4  0x00000000007d390c in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbSceneTree::*)()>::call(void (WbSceneTree::*)(), WbSceneTree*, void**) (
    f=(void (WbSceneTree::*)(WbSceneTree * const)) 0x588b74 <WbSceneTree::handleDoubleClickOrEnterPress()>, o=0x19374730, arg=0xf31bc90)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
#5  0x00000000007fcc8f in QtPrivate::FunctionPointer<void (WbSceneTree::*)()>::call<QtPrivate::List<>, void>(void (WbSceneTree::*)(), WbSceneTree*, void**) (
    f=(void (WbSceneTree::*)(WbSceneTree * const)) 0x588b74 <WbSceneTree::handleDoubleClickOrEnterPress()>, o=0x19374730, arg=0xf31bc90)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
#6  0x00000000007e1f08 in QtPrivate::QSlotObject<void (WbSceneTree::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x345a7640, r=0x19374730, a=0xf31bc90, ret=0x0)
    at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:418
#7  0x0000000012b6b40c in void doActivate<false>(QObject*, int, void**) ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#8  0x00000000006d5769 in WbTreeView::doubleClickOrEnterPressed (
    this=0x2ce1afd0) at build/debug/WbTreeView.moc.cpp:227
#9  0x00000000005f496b in WbTreeView::mouseDoubleClickEvent (this=0x2ce1afd0,
    event=0xf31c3a0) at scene_tree/WbTreeView.cpp:154
#10 0x0000000066247b61 in QWidget::event(QEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#11 0x00000000662ee884 in QFrame::event(QEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#12 0x0000000012a80b0c in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) [clone .part.0] () from D:\msys64\mingw64\bin\Qt5Core.dll
#13 0x0000000066207f92 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from D:\msys64\mingw64\bin\Qt5Widgets.dll
#14 0x000000006620f427 in QApplication::notify(QObject*, QEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#15 0x0000000012a82300 in QCoreApplication::sendSpontaneousEvent(QObject*, QEvent*) () from D:\msys64\mingw64\bin\Qt5Core.dll
#16 0x000000006620e327 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#17 0x000000006625f93b in QWidgetWindow::handleMouseEvent(QMouseEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#18 0x0000000066262a92 in QWidgetWindow::event(QEvent*) ()
   from D:\msys64\mingw64\bin\Qt5Widgets.dll
#19 0x0000000066207fa3 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from D:\msys64\mingw64\bin\Qt5Widgets.dll
#20 0x0000000012a82300 in QCoreApplication::sendSpontaneousEvent(QObject*, QEvent*) () from D:\msys64\mingw64\bin\Qt5Core.dll
#21 0x00000000122300ef in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from D:\msys64\mingw64\bin\Qt5Gui.dll
#22 0x000000001220552c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from D:\msys64\mingw64\bin\Qt5Gui.dll
#23 0x0000000012ade0fe in QEventDispatcherWin32::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from D:\msys64\mingw64\bin\Qt5Core.dll
#24 0x000000006dd40bb5 in QWindowsGuiEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from D:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll
#25 0x0000000012a803a3 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from D:\msys64\mingw64\bin\Qt5Core.dll
#26 0x0000000012a89047 in QCoreApplication::exec() ()
   from D:\msys64\mingw64\bin\Qt5Core.dll
#27 0x00000000004c0da5 in WbGuiApplication::exec (this=0xf31fcc0)
    at gui/WbGuiApplication.cpp:273
#28 0x000000000063ab9c in main (argc=1, argv=0x15213f80) at gui/main.cpp:207
```

omichel, 2020-06-23:
In fact the crash occurs with any world, by double-clicking on any node of the scene tree.

omichel, 2020-06-23:
Strangely, this fixes the crash:
```diff
--- a/src/webots/vrml/WbNode.hpp
+++ b/src/webots/vrml/WbNode.hpp
@@ -331,10 +331,9 @@ private slots:
 private:
   WbNode &operator=(const WbNode &);  // non copyable

-  QString mUrdfPrefix;
-
   // for all nodes
   WbNode *mParent;
+  QString mUrdfPrefix;
   WbNodeModel *mModel;
   int mUniqueId;
```
I am puzzled...

DavidMansolino, 2020-06-23:
:astonished: 

`mUrdfPrefix` is not initialized in the constructor, but I doubt this is the case.

omichel, 2020-06-23:
I tried to initialize it in the constructor, this doesn't change anything.

omichel, 2020-06-24:
Fixed in #1824.

PR #1824 (2020-06-24): Fixes crash when clicking on a node in the scene tree
Fixes #1821.

This simple fix is extremely strange and I don't understand it.
But it fixes #1821.

I isolated the problem by testing the commit history.
closed 2020-06-24 by omichel