cyberbotics/webots #1841: motor minPosition/maxPosition gets reset, if range doesn't include 0

Simon-Steinmann, 2020-06-27:
**Describe the Bug**
The issue arises when we define a Rotational motor like this:
```
RotationalMotor {
                  name "finger_2_joint_1"
                  minPosition 0.0495
                  maxPosition 1.2218
}
```
or this:
```
RotationalMotor {
                              name "finger_2_joint_3"
                              minPosition -1.2217
                              maxPosition -0.0523
                            }
```
In the first case, minPosition gets set to 0, and in the second case, maxPosition gets set to 0.

**Steps to Reproduce**
1. Change the minPosition and maxPosition of a rotational motor to either both positive or both negative in a proto file
2. start the simulation with a <void> controller and open the robot window
3. check the limits of the Rotational Motor

**Expected behavior**
Either the minPosition or the maxPosition will be set to 0

**Additional context**
Many robots, especially when using manufacturer's provided urdf files, define joint limits in a range above or below 0. 

I believe the problem to be here:
https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbMotor.cpp
line 178-180

Simon-Steinmann, 2020-06-27:
My cpp skills are very limited, so I wont try to provide a solution, but I think it should set the position to either the min or max position (whichever is closer to 0)

DavidMansolino, 2020-06-29:
Actually, if you want to have a minPosition that is bigger than 0, you should first move the joint to this position (or bigger) using the jointParameters.
But you are right, we should improve the warning message to explain this, or even better change automatically the joint position in this case.

Simon-Steinmann, 2020-06-29:
The thing is, if you want to use provided files from manufacturers, and perhaps use their controllers, it would be best if the definitions are the same. I found this issue when trying to set the correct joint limits to the 3f-robotiq gripper (they are wrong in the one included in webots).  Setting an explicit motor position would be different for the simulation and the real robot. That is an issue.

DavidMansolino, 2020-06-29:
I am not sure to understand what is the issue with setting the joint position in the joint parameters.
You want to set the limit and have the current position to be out of the range limit?

Simon-Steinmann, 2020-07-06:
@DavidMansolino 
The issue is, that we cannot simply import a urdf with the limits set by the manufacturer, without meticulous adjustment of the model. I assume that they did not rotate the joint to keep the initial frames in the same rotation and rather use the position limit to achieve that task. In the end it should not matter which approach one chooses in my opinion. Requiring that joint positions MUST include 0 is arbitrary in my opinion, as all positions are relative anyways. But I might be missing something. 

If you decide to keep it the way it is (and perhaps adding a more detailed error description), it would be nice to have a tutorial on how to adjust the model correctly. For example, how would this have to be changed?:
```
 endPoint Solid {
          translation -0.0455 0.0214 0.036
          rotation 0.7068252250609627 0.707388225240228 3.0000009552334565e-06 3.141598
          children [
            HingeJoint {
              jointParameters HingeJointParameters {
                axis 0 0 1
                anchor 0.02 0 0
              }
              device [
                RotationalMotor {
                  name "finger_1_joint_1"
                  maxVelocity 100.0
                  minPosition 0.0495
                  maxPosition 1.2218
                  maxTorque 100.0
                }
```

DavidMansolino, 2020-07-07:
Have you tried with something like this:
```
        endPoint Solid {
          translation -0.0455 0.0214 0.036
          rotation 0.7068252250609627 0.707388225240228 3.0000009552334565e-06 3.141598
          children [
            HingeJoint {
              jointParameters HingeJointParameters {
                axis 0 0 1
                anchor 0.02 0 0
                position 0.0495      # CHANGE HERE
              }
              device [
                RotationalMotor {
                  name "finger_1_joint_1"
                  maxVelocity 100.0
                  minPosition 0.0495
                  maxPosition 1.2218
                  maxTorque 100.0
                }
```

Simon-Steinmann, 2020-07-07:
Is position a valid field for HingeJointParameters? It is not listed in the documentation 

I have changed it like this:
```
 endPoint Solid {
          translation -0.0455 0.0214 0.036
          rotation 0.7068252250609627 0.707388225240228 3.0000009552334565e-06 3.141598
          children [
            HingeJoint {
              jointParameters HingeJointParameters {
                axis 0 0 1
                anchor 0.02 0 0
              }
              device [
                RotationalMotor {
                  name "finger_1_joint_1"
                  maxVelocity 100.0
                  maxPosition 1.1723    # CHANGE HERE
                  maxTorque 100.0
                }
                PositionSensor {
                  name "finger_1_joint_1_sensor"
                }
              ]
              endPoint Solid {
                translation 0.02 0 0
                rotation 0.0 0.0 1.0 0.0495  # CHANGE HERE
```

omichel, 2020-11-30:
I am closing this old stalled issue. Feel free to re-open if needed.

closed 2020-11-30 by omichel

Simon-Steinmann, 2020-11-30:
I think we adressed this in the urdf2webots converter, so should be fine for most cases.