cyberbotics/webots #1843: Broken Backward compatibility with ControllerArgs

DavidMansolino, 2020-06-29:
**Describe the Bug**
In some cases (e.g. procedural PROTOs) the change of the `controllerArgs` fields from SFString to MFString broke the backward compatibility.

For example, with such PROTO:
```
#VRML_SIM R2020a utf8
# tags: static

PROTO yCar [
  field SFVec3f      translation              0 0.4445 0
  field SFRotation   rotation                 0 1 0 0
  field SFColor      color                    0.92 0.94 0.94
  field SFString     controller               "void"
  field SFString     controllerArgs           ""
]
{
  Car {
    translation IS translation
    rotation IS rotation
    controller IS controller
    controllerArgs IS controllerArgs
    recognitionColors [
      %{= fields.color.value.r }% %{= fields.color.value.g }% %{= fields.color.value.b }%
    ]
  }
}
```

2 controllerArgs fields are generated in the scene-tree (without any warnings):
[image]

None of them is usable (not passed to the controller).

omichel, 2020-06-29:
I believe this is more a general PROTO checking issue: this [method](https://github.com/cyberbotics/webots/blob/master/src/webots/vrml/WbProtoModel.cpp#L457) is not called on the MyCar device (it should check the match of the IS data types).

omichel, 2020-07-01:
Fixed in #1847

PR #1847 (2020-07-01): Fix warning for is datatype mismatch in procedural protos
Fixes #1843.

This PR:
- removes some useless piece of code: we never need to write the IS keyword in a `.wbt` file.
- fixes missing warning about IS datatype mismatch for procedural protos.

It doesn't automatically fixes the backwards compatibility problem related to the controllerArgs datatype change (which should be nevertheless relatively rare).
closed 2020-07-01 by omichel