cyberbotics/webots #1849: touch sensor didn't work in webots_ros (complete_test example) and how to add touch sensor for gripper?

ipa-rwu, 2020-06-29:
Hello,

I want to add a touchsensor at the endpoint of gripper. But it doesn't work.
So I tried the example complete_test in webots_ros package. Basiclly, what I did is:

- start the world: complete_test.wbt
- put a object next to touch sensor
- call touchsensor service: `rosservice call /my_robot/touch_sensor/enable "value: 1"`
- Monitoring touchsensor topic: `rostopic echo /my_robot/touch_sensor/value`

I also tried:
- `roslaunch webots_ros complete_test.launch `
- and monitoring touchsensor topic: `rostopic echo /my_robot/touch_sensor/value`

I checked the document [https://cyberbotics.com/doc/reference/touchsensor?tab-language=c++#collision-detection](url)
I am wondering if if is because of "boundingObject" in touchsensor is null. As you can see this screenshot
[image]

What I tried on gripper is:
[image]
But then it seems gripper can't work properly.  Are there any examples of how to add a touch sensor?

Thanks in advance.

omichel, 2020-06-30:
Can you please upload a ZIP including your world file and proto file so that I can try to reproduce the problem?

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel