cyberbotics/webots #1853: Crash When Changing ImageTexture.filtering field

DavidMansolino, 2020-06-30:
**Describe the Bug**
Changing the `filtering` field to a negative value in the following ImageTexture node:
```
Floor {
  appearance PBRAppearance {
    baseColorMap ImageTexture {
      url [
        "textures/checkered_marble.jpg"
      ]
    }
  }
}
```

Makes Webots crash with the following stack (on an up to date develop branch):
```
0x000055d06e90ab78 in WbTreeItem::parent (this=0x1000000010000)
    at scene_tree/WbTreeItem.hpp:42
42	  WbTreeItem *parent() const { return mParent; }
(gdb) where
#0  0x000055d06e90ab78 in WbTreeItem::parent() const (this=0x1000000010000)
    at scene_tree/WbTreeItem.hpp:42
#1  0x000055d06e90ecdf in WbSceneTreeModel::itemToIndex(WbTreeItem const*) const (this=0x55d091a32c10, item=0x7f3db400b2c0)
    at scene_tree/WbSceneTreeModel.cpp:210
#2  0x000055d06e909323 in WbSceneTree::applyNodeRegeneration(WbNode*)
    (this=0x55d07056d810, node=0x55d09269da50)
    at scene_tree/WbSceneTree.cpp:1311
``

stefaniapedrazzi, 2020-07-02:
I tried to reproduce this issue on the develop branch and I get a crash but with a different stack.
```
Thread 1 "webots-bin" received signal SIGSEGV, Segmentation fault.
0x00007ffff305740d in __dynamic_cast () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff305740d in __dynamic_cast () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x0000555555a78f55 in WbPbrAppearance::roughnessMap() const (this=0x55556158eb40) at nodes/WbPbrAppearance.cpp:307
#2  0x0000555555a77842 in WbPbrAppearance::preFinalize() (this=0x55556158eb40) at nodes/WbPbrAppearance.cpp:87
#3  0x0000555555aedeb0 in WbShape::preFinalize() (this=0x7fffd800bb80) at nodes/WbShape.cpp:69
#4  0x00005555559c48aa in WbGroup::preFinalize() (this=0x555561581e40) at nodes/WbGroup.cpp:67
#5  0x0000555555b79f06 in WbTransform::preFinalize() (this=0x555561581e40) at nodes/WbTransform.cpp:56
#6  0x0000555555b12898 in WbSolid::preFinalize() (this=0x555561581e40) at nodes/WbSolid.cpp:273
#7  0x000055555592bbef in WbBaseNode::finalize() (this=0x555561581e40) at nodes/WbBaseNode.cpp:85
#8  0x0000555555b5d7c0 in WbTemplateManager::regenerateNode(WbNode*) (this=0x555556b42ab0, node=0x55555c1fd290) at nodes/utils/WbTemplateManager.cpp:399
#9  0x0000555555b5c628 in WbTemplateManager::regenerateNodeFromField(WbNode*, WbField*, bool) (this=0x555556b42ab0, templateNode=0x55555c1fd290, field=0x55555c1a0340, isParameter=false)
    at nodes/utils/WbTemplateManager.cpp:205
#10 0x0000555555b5c467 in WbTemplateManager::regenerateNodeFromFieldChange(WbField*) (this=0x555556b42ab0, field=0x55555c1a0340) at nodes/utils/WbTemplateManager.cpp:177
```

So it could be that part of the issue is fixed in  #1868, but there are still some other issues to solve.
In any case the source of the crash is always related to template PROTO regeneration (and testing I'm finding many other related issues).

stefaniapedrazzi, 2020-07-06:
In this particular case, the crash seems to be due to a nested PROTO regeneration triggered by the ``WbImageTexture::updateFiltering`` function when finalizing the newly generated node:
https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbImageTexture.cpp#L220

closed 2020-07-22 by stefaniapedrazzi