cyberbotics/webots #1855: SliderJoint/LinearMotor endPoint moves when the simulation starts (no controller)

Acwok, 2020-06-30:
**Describe the Bug**
The endPoint position of the SliderJoint moves when the simulation starts.

**Steps to Reproduce**
- Create a robot with a 2m SliderJoint and a linearMotor
- Define the joint and motor limits [0 1.9]
- Use a 0.1m solid as endPoint
- Move the solid to the left of the red solid

Here is a simple world showing the issue: [rail.wbt.zip](https://github.com/cyberbotics/webots/files/4851173/rail.wbt.zip)
Here is a video showing the issue: https://youtu.be/3RCH9gzHMFI

**Expected behavior**
The white cube (endPoint Solid) don't move as there is no controller nor external forces

**System**
- Ubuntu 18.04 in a docker with GPU acceleration
- Webots build from sources commit fe5893b968e160447a52cc4619ad68a8f41f8022

DavidMansolino, 2020-06-30:
How heavy is you Solid node?
Would it be possible that the motor is not strong enough to maintain it in place?

DavidMansolino, 2020-06-30:
I checked your simulation in detail and found the reason of what you are observing.
The problem is coming from the joint limit, since you defined the limit (in the JointParameters node) to be [0, 1.9] and your initial position is 0.0, the joint is in a limit case, since it touches the limit it is pushed away (quite strongly, probably because there is no friction or gravity) and then the motor is pushing it back to the 0 position (which is it's default target position), it finally stabilize to a position extremely close to 0 (e.g. 1e-05).
To avoid this you can either:
  - Set the start position to be slightly bigger than 0 (e.g. 1e-05).
  - Set the minStop to be slightly smaller than 0 (e.g. -1e-05).
  - Instead of using the joint limit (minStop and maxStop) use the motor limit (minPosition and maxPosition) only, they are way more stable and in most of the case sufficient (they might be useful only if you set the torque/force of the motor to a very low value).

Thank you for the dockerfile, that looks very good, I will definitely have a try one of these days.
I am by the way planning to create automatic docker images of Webots using Github Action on a daily basis, would you be interested by such image?

Acwok, 2020-07-01:
Thank you! Indeed, it is working if I remove the minStop and maxStop fields.  
It is also working if I set minStop=-0.00001 and maxStop=5.00001 but if I understand correctly it is better not to set them.

I also noticed some instabilities when adding the third joint of my robot.  
The robot pose is slightly moving during the very first second of the simulation.  
It always moves for exactly 1s and then the bounding boxes turn to blue.   
See https://youtu.be/z2AsC4mDZGo.
Here is the corresponding world: [unstable_robot.wbt.zip](https://github.com/cyberbotics/webots/files/4856144/unstable_robot.wbt.zip)  

If I remove the third joint, the robot is not moving.
See https://youtu.be/z2AsC4mDZGo?t=37
Here is the corresponding world: [stable_robot.wbt.zip](https://github.com/cyberbotics/webots/files/4856147/stable_robot.wbt.zip)  

I've tried moving the last endSolid position (to avoid any collision), changing its physics parameters, but without success.  
I really don't understand what it could be. Do you have any idea ?
Also, what is this 1s step ? it happens in the stable robot too.

> I am by the way planning to create automatic docker images of Webots using Github Action on a daily basis, would you be interested by such image?

In the H2020-PULSAR context, we need to have a stable image. So, when I see new interesting features in Webots, I always make another image and ensure it doesn't introduce new issues. We also test the image on different hardware before switching.  
However, having Webots in a docker is great for sharing simulations and ensuring that the person who is testing your simulation has the same environment as you. And unlike virtual machines, you still have very good performances.  
I think it could interest Webots users to have a docker image available on new Releases.

Acwok, 2020-07-01:
I noticed something that might be linked: https://youtu.be/Xo6V7gqOWXI  
if I add the stable_robot to the children field of another robot (the green cube in the video), everything stays stable, I mean neither the green cube nor the arm positions move during the first second of the simulation.  
But if I rotate the arm by 90° the small movements appear.
Here is the world file if you want to take a look: [stable_robot_in_robot.wbt.zip](https://github.com/cyberbotics/webots/files/4857258/stable_robot_in_robot.wbt.zip)

DavidMansolino, 2020-07-02:
This is most probably due to some minor instabilities (caused by rounding issues or similar), but I can see that they are extremely small (e.g. `e-20`) and could therefore be neglected.

The fact that they stop after 1 second is because Webots has a mechanism called the sleep mechanism that puts an object to sleep if it almost didn't moved for a period of time. The motion and time threshold can be defined in the WorldInfo node, thanks to the `physicsDisableTime` (1 second by default) , `physicsDisableLinearThreshold` and `physicsDisableAngularThreshold` fields: https://www.cyberbotics.com/doc/reference/worldinfo

Acwok, 2020-07-03:
Indeed it can be neglected.

DavidMansolino, 2020-07-14:
Original issue solved

closed 2020-07-14 by DavidMansolino

DavidMansolino, 2020-08-12:
> The solid is 10Kg and the red rail is 100Kg. Motor maxForce is 1000N,
> I have tried with 10000N and it is still moving (almost teleporting).
> Changing the acceleration value or the PID only affects the time it takes to get back to position 0.
> I forgot to mention that gravity is set to 0.
> 
> > P.S: I am quite interested about your docker with GPU acceleration in which you run Webots, do you have a Dockerfile to build the image that you could share?
> 
> Sure, we've made this docker for the dISAS demonstrator of the [H2020-PULSAR](https://www.h2020-pulsar.eu/) project.
> Here is the dockerfile: [Dockerfile.zip](https://github.com/cyberbotics/webots/files/4852273/Dockerfile.zip).
> If you have any question feel free to ask, we are using docker a lot at [Magellium](https://www.magellium.com/fr/).

@Acwok thank you again for sharing the Dockerfile, I tried it, it 'compiles' fine but there are some issues when running Webots from the container:
```
Warning: could not connect to display 
Info: Could not load the Qt platform plugin "xcb" in "/usr/local/webots/lib/webots/qt/plugins" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.
```

I see that from the docker you are not exporting the `DISPLAY` environment variable or using any virtual framebuffer (such as xvfb) which can explain this error. Did you experienced something similar?

Acwok, 2020-09-03:
@DavidMansolino  
Sorry for the late reply, I missed the notification.  
You need to enable connections to server x using: `xhost +local:root` before running a container.
Here is the script we use to launch our image: [launcher.zip](https://github.com/cyberbotics/webots/files/5169075/launcher.zip)

DavidMansolino, 2020-09-07:
@Acwok thank you I have know been able to make it work :smiley: