cyberbotics/webots #1867: Improve detection of PROTO template regeneration fields

stefaniapedrazzi, 2020-07-02:
Are there other ways to access a parameter from a Lua statement other than using the `fields.` prefix?

Otherwise we could maybe improve the search by including the `fields.` prefix in the regular expression, something like
```
QRegExp(QString("(^|[^a-zA-Z0-9_])fields\.%1($|[^a-zA-Z0-9_])").arg(QRegExp::escape(model->name())))))
```

Here is the current regular expression used to extract if a PROTO field triggers template regeneration:
https://github.com/cyberbotics/webots/blob/0fe9e56f266a8855e9ab74e811f128f083307e5b/src/webots/vrml/WbProtoModel.cpp#L204

DavidMansolino, 2020-07-02:
No, I don't see any other way to get the value.
But we can have something like this `fields.wheelFrontRight.value.fields.tireRadius.value` with multiple time the `fields.` keyword, but this should not cause problem.

DavidMansolino, 2020-07-02:
Oh, just thought that one tricky way could be to copy the dictionnary:
```
local newDict = fields
local parameterValue = fields.myField.value
``
But none of our examples are using this, and I don't see any reason for doing this.

omichel, 2020-07-03:
So, we should probably search for `fields` instead of `fields.` to be on the safe side.

DavidMansolino, 2020-07-03:
I am not sure to understand how this will help to determine if the field is regenerative or not?

omichel, 2020-07-03:
Sorry, forget about my comment. I was confused. The only way to handle this rare case would be to search for this kind of statement (like `newDict = fields`, `newDict2=fields`) and add all the `newDict.` (or whatever is found), `newDict2.`, etc. in the search pattern in addition to `fields.`. But I am not sure it is worth adding that complexity to our code for such a rare use case.

closed 2020-08-10 by stefaniapedrazzi