cyberbotics/webots #1870: Webots icon is not shown after installing in linux

yashpatel97, 2020-07-02:
**Describe the Bug**
I installed webots R2020a-rev on Ubuntu Linux (version -18.04). Webots was installed using Debian Package with the Advanced Packaging Tool (APT) as per the instructions provided on the following link --https://cyberbotics.com/doc/guide/installation-procedure#installing-the-debian-package-with-the-advanced-packaging-tool-apt. But after the installation, the webots icon does not appear on the webots application.

**Steps to Reproduce**
1. Go to 'https://cyberbotics.com/doc/guide/installation-procedure'
2. Click on 'Installing the Debian Package with the Advanced Packaging Tool (APT)'
3. follow the commands given on the linux terminal for webots installation.
4. After the installation is completed check the icon of webots application.

**Expected behavior**
The webots application seems to be working fine when using it. But the official icon of webots is not displayed. 

**Screenshots**
The screen shot of the issue is also provided that explains the issue.
[image]

**System**
 - Operating System: [Linux Ubuntu 18.04,- dual booted with windows 10]
 - Graphics Card: [NVIDIA GeForce GTX 1050Ti]

stefaniapedrazzi, 2020-07-02:
This should already be fixed in the latest nightly build (see  #1607 and  #1721).
Could you please try installing the latest nightly build and let us know if you still have the issue with the icon?
https://github.com/cyberbotics/webots/releases/download/nightly_1_7_2020/webots_2020a-rev2_amd64.deb

yashpatel97, 2020-07-02:
> This should already be fixed in the latest nightly build (see #1607 and #1721).
> Could you please try installing the latest nightly build and let us know if you still have the issue with the icon?
> https://github.com/cyberbotics/webots/releases/download/nightly_1_7_2020/webots_2020a-rev2_amd64.deb

okay, will do that ! But do I have to delete the previous version and then proceed with the installation of the latest release ?

DavidMansolino, 2020-07-03:
Yes, it is better to remove the previous one before.

yashpatel97, 2020-07-03:
> Yes, it is better to remove the previous one before.

thank you, the issue is resolved after installing the latest release with link provided !

closed 2020-07-03 by omichel