cyberbotics/webots #1871: Camera image nondeterministic across graphics cards

MaxGaukler, 2020-07-03:
**Describe the Bug**
We are using Webots for a student competition and experience that the simulated camera image depends on the graphics card and possibly even the driver version. The difference is especially pronounced for software OpenGL rendering, but also exists on other systems. I do understand that software rendering is not officially supported, however this is problematic if you want to provide a virtual image or server as reference environment. One example with important industrial application are Continuous Integration tests on some cloud service that will either not have a GPU or may have varying GPU vendors.

The [documentation](https://cyberbotics.com/doc/guide/modeling#how-to-make-replicabledeterministic-simulations) claims that simulations are replicable (under some conditions) on the same OS and platform (which I interpret as "Webots on Ubuntu version x.y on a 64bit processor will always behave the same, no matter on which PC")

**Steps to Reproduce**
- Settings:
  - Startup Mode = Pause, Threads = 1
  - OpenGL: Everything to disabled/lowest.
- Open sample project: samples/devices/worlds/camera.wbt
- Robot - Children - Camera -> Noise =0, Antialiasing=False

- Run two simulation steps (Ctrl-1)
- Hover over camera image and read pixel values in status bar.
- Compare for two different systems. The error seems to be largest where the texture has sharp lines.
  - pixel(17,47)=  NVIDIA: #BC8876  VirtualBox SoftwareGL: #B8816D
  - pixel(70,40)=  NVIDIA: #157215  VirtualBox SoftwareGL: #157215

In more complicated examples, it seems that the SoftwareGL rendering is blurred by a few pixels. We had similar (though less pronounced) differences between different Windows 10 systems and even different GPU driver versions. As a result, controller behavior is not reproducible over different systems or if an automatic update occurs.

**Expected behavior**
My expectation would be that like in other simulators, the same input produces the same output on any host system, up to minor numerical noise. Instead, it seems to highly depend on the OpenGL implementation of texture rendering (and settings?). 

**Screenshots**

[image]

**System 1: NVIDIA**
 - Operating System: Linux Ubuntu 18.04.4 with nvidia-driver-435
 - Graphics Card: NVIDIA Corporation GP107GLM [Quadro P1000 Mobile]

**System 2: VirtualBox Software GL**
 - Operating System: Linux Ubuntu 18.04.4 inside VirtualBox Version 5.2.34_Ubuntu r133883, running on host System 1
 - Graphics Card: VirtualBox, all acceleration disabled

**Additional context**
Here's a more complicated example from https://gitlab.cs.fau.de/ak22emur/mechsys-webots illustrating the real-world impact of the problem: A robot uses a 1x30 pixel camera as a "barcode scanner" to read a pattern from the floor. Threshold-based algorithms behaved quite differently on different systems, and even across different GPU driver versions on the same Windows 10 system.

[image]

omichel, 2020-07-03:
Thank you for this detailed report.
Unfortunately, the OpenGL rendering is depending on the OpenGL vendor implementation (NVIDIA, AMD, Intel, Mesa).
That means that the rendering may have slight differences as the ones you observed when running Webots on different operating systems or different graphics cards or even different version of a graphics drivers.
I am afraid this cannot be fixed.
If you need pixel-level reproducibility for robot cameras you should either:
- specify a hardware & software configuration: operating system + graphics card + version of the graphics card driver.
- rely on software-only rendering with a specific version of Mesa, disabling explicitly hardware acceleration (but this will run Webots slowly).

We are going to improve the documentation with respect to this.

**Note:** in our CI tests, we introduced some tolerance on pixel value measurement, so that our CI tests are passing on various OSes / graphics card configurations. See for example [this](https://github.com/cyberbotics/webots/blob/master/tests/api/controllers/camera_image_update/camera_image_update.c#L27-L29).

omichel, 2020-07-17:
Fixed in #1932.

PR #1932 (2020-07-17): Fixed reproducibility documentation
Fixes #1871.
closed 2020-07-17 by omichel