cyberbotics/webots #1872: Large variation in simulation speed between multiple launches

Acwok, 2020-07-03:
**Describe the Bug**
The simulation speed is not always the same between consecutive runs.  
It makes it difficult to see if the modifications we are making improve the performances or not.  
Here are some examples:  
- video 1: https://youtu.be/eYst9KKvWjM corresponding world: [speed_variability_tile.wbt.zip](https://github.com/cyberbotics/webots/files/4868895/speed_variability_tile.wbt.zip)
- video 2: https://youtu.be/msWvM3OhkvQ corresponding world: [speed_variability_tile_base.wbt.zip](https://github.com/cyberbotics/webots/files/4868902/speed_variability_tile_base.wbt.zip)  
- video 3: https://youtu.be/_BuD-p3GcKI corresponding world: [speed_variability.wbt.zip](https://github.com/cyberbotics/webots/files/4868910/speed_variability.wbt.zip)

In these examples:
- randomSeed was set to 1
- optimalThreadCount was set to 1
- tools/preferences/NumerOfThreads was set to 4
- basicTimeStep was set to 8ms
- all controllers fields were set to ""

**Expected behavior**
The simulation speed should be almost the same between two runs if no other applications are running in background and consuming system resources.

**System**
- Ubuntu 18.04 in a docker with GPU acceleration
- Webots build from sources commit 65e5a4919c6c39694df4ffea9a54cf04a6e97302

omichel, 2020-07-03:
Are you sure the difference is coming from Webots? It may also be due to the controllers that run at different speeds during different runs due to possibly networking, variable CPU load, race conditions, memory allocations causing swap operations, etc.

Also, beware that the caching of library loading, memory cache, memory/disk swap, background processes, etc. may cause several runs to execute at significantly different speeds.

To better understand the problem, you should try to determine if this speed variation is coming from Webots or from your robot or supervisor controllers. You should also monitor the other processes running on the machine. Although I don't know them very well and cannot make any recommendation, there are certainly monitoring tools on Linux that will help you with these investigations.

DavidMansolino, 2020-07-03:
I can indeed reproduce this on a native (non-docker environment), from one run to the next one the speed difference can reach up to 20x which is quite big (small variations are unavoidable due to fluctuations of the computer load, but such big, it looks indeed like a bug).

> It may also be due to the controllers that run at different speeds during different runs due to possibly networking, variable CPU load, race conditions, memory allocations causing swap operations, etc.

It is unfortunately not the controller as this world is not using any controller.

I also checked if the computer load is changing, and it doesn't seem to be the case (except Webots, the CPU is used < 3% and there is plenty of free memory).

omichel, 2020-08-26:
I believe this bug may have been fixed in #1994.
Can you try a recent [nightly build](https://github.com/cyberbotics/webots/releases) and confirm that it is now fixed?

Acwok, 2020-11-05:
@omichel 
The wbt files I had provided are not working anymore, instead I've tested with the world  provided here https://github.com/cyberbotics/webots/issues/2426. 

If I restart Webots, the simulation speed is the same. 
However if I only reload the world, (as I was doing in the videos above) the simulation speed drops by almost 1x each time. 
It seems that the large variations in simulation speed are not random anymore.

I've tested with Webots built from fix-performance-drop and R2020b, same result.

omichel, 2020-11-05:
I cannot reproduce this on my Windows 10 machine. Reloading the simulation several times doesn't change the simulation speed.

Acwok, 2020-11-05:
I may have exaggerated a bit about the "almost 1x", but you can see the phenomenon in this video: https://youtu.be/nrUwuJeXd1k
It starts at almost 9x and quickly drops to 6x after some reloads.
The last three run were made after restarting webots, and the speed always get back to almost 9x.

It is not an issue for my usecase, I just wanted to let you know.
The original issue is fixed to me.

closed 2020-11-23 by omichel