cyberbotics/webots #1874: Community Model Repo

DavidMansolino, 2020-07-03:
Many users are not publishing their models/controllers because our quality assurance and quality standard are quite high.

We have plans for allowing the users to share their model, but these are long term plans. In the meantime, a simple solution would be to create a 'webots-coomunity' (or anything similar) repo where users could add their models/controllers/simulations with a lower quality standard and without too detailed review.

Simon-Steinmann, 2020-07-03:
This would be great. Almost a year ago, I tried out webots as an alternative to Gazebo, but ultimately dropped, because I didn't manage to implement the jaco2 kinova arm I was working on. Even with using the urdf converter. 
For the future implemented feature, I attached a image, of how it could look like[
[image]
](url)

DavidMansolino, 2020-07-03:
Thank you for the feedback!

About the visualization, we have already a similar feature, allowing to add a new path to search for PROTO/controllers, etc.
If you go to the preferences, it is called 'Extra projects path'.

DavidMansolino, 2020-07-14:
This is now live at: https://github.com/cyberbotics/community-projects

closed 2020-07-14 by DavidMansolino