cyberbotics/webots #1881: Supervisor getFromDef() produce distinct errors

martinpla, 2020-07-07:
**Describe the Bug**
When using Supervisor getFromDef() to get a PROTO Node I am getting two errors:
1) Sometimes "abort: api/robot.c:347: assertion failed: tag < robot.n_device"
2) Sometimes "A fatal error has been detected by the Java Runtime Environment: #  EXCEPTION_ACCESS_VIOLATION "
which produces a hs_error_pid file

**Steps to Reproduce**
1. Download [project ]( https://github.com/martinpla/foraging_project)
2. Run project
3. Position robot on FoodSource (green circle). After loading food it will return to the nest (blue circle), leaving behind a trail of pheromone (yellow cells). Cell is a PROTO that I defined. You will see DEF Cells in the scene tree.
4. After unloading the food and when positioned on the trail, the robot will use Supervisor methods to look for cells with field "value" positive. ( See method **pheromoneValue()** from **MyRobot.java** )
5. After some time of the robot being on the trail, error will happen.

**Expected behavior**
I expect the robot to look for pheromone on his current cell or the adyacent cells. FollowTrail.java implements this behavior, takeControl() method looks for pheromone nearby, action() method selects one cell and moves to it.
Both methods call MyRobot.pheromoneValue() which uses Supervisor.getFromDef() (line 151 of MyRobot.java) and other Supervisor functions. 
When using any of these, error takes place (see attached error files )
[hs_err_pid6168.log](https://github.com/cyberbotics/webots/files/4885454/hs_err_pid6168.log)
[hs_err_pid8420.log](https://github.com/cyberbotics/webots/files/4885456/hs_err_pid8420.log)
[hs_err_pid13404.log](https://github.com/cyberbotics/webots/files/4885460/hs_err_pid13404.log)

`public int pheromoneValue(int x, int y) {
//look for cell x y
Node cell = this.getFromDef(String.format("cell_%d_%d", x, y));
if(cell == null) { return 0;}
else { Field f = cell.getField("value");
       if(f == null) { return 0;} 
       else { int value = f.getSFInt32(); 
              return value;
             } 
       }
}          
`
**Screenshots**
[image]
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: [Intel(R) HD Graphics 5500]

**Additional context**
The controller uses Subsumption architecture ( package lejos.robotics.subsumtion ). Lejos.subsumption provides:
- class Arbitrator:  selects which behavior will take control of the robot
- interface Behavior: specifies a simple behavior with three methods: takeControl(), action() and suppress().
[link](http://www.lejos.org/nxt/nxj/tutorial/Behaviors/BehaviorProgramming.htm#2)

Take into account that Arbitrator creates a second thread that keeps asking for behaviors that want to take control of the robot.

DavidMansolino, 2020-07-08:
@martinpla thank you for this bug report.
Which version of Webots are you using? Did you compiled Webots yourself?
Your simulation and controller are quite complex, it is therefore quite difficult to isolate the problem.
Can you try reproducing it in a simpler setup?

martinpla, 2020-07-08:
Hello David, thank you for your response. I am using version R2020a. I did not compile it myself, I simply installed it via the Windows Installer.
Yesterday I tried installing Webots on Ubuntu (same PC) and the same two errors happened.
I observed that not only the method getFromDef() is giving trouble, also : 
* importNodeFromString() (see line 86 of hs_err_pid6168.log)
*getSFInt32()  ( see line 91 of hs_err_pid8420.log)
*or even Robot.step() see line 86 of [hs_err_pid13188.log](https://github.com/martinpla/foraging_project/blob/master/controllers/ForagingControllerKhepera/hs_err_pid13188.log)

I have cleaned the code a bit in my github repository and I will be trying to reproduce the error in a simpler setup.

PD: I have been thinking about something that could be related to error 2 - EXCEPTION_ACCESS_VIOLATION.
As I told you I am using Arbitrator and Behavior classes from lejos.robotics.subsumption package. 
Arbitrator creates a second threadappart from the main thread. 
Maybe this thread is provoking the access violation error

DavidMansolino, 2020-07-09:
That can indeed be an explanation, but usually, the Webots API is thread-safe so it should not cause problem.
Also, are you importing new nodes in the robot itself (or another robot)?

DavidMansolino, 2020-08-27:
@martinpla we fixed recently several issues related to node importation, can you please re-try using a nightly build of Webots R2020b-rev1: https://github.com/cyberbotics/webots/releases

Thank you.

omichel, 2020-11-30:
I am closing this stalled issue as we had no answer from @martinpla. Feel free to re-open if needed.

closed 2020-11-30 by omichel