cyberbotics/webots #1882: Crash when editing baseColorMap url

MaxGaukler, 2020-07-07:
**Describe the Bug**
Webots often crashes when modifying the appearance of a RectangleArena. Editing the texture image path in the text field seems especially problematic.

**Steps to Reproduce**
Original case:
1. `git clone https://gitlab.cs.fau.de/ak22emur/mechsys-webots`
2. open worlds/simulation.wbt
3. RectangleArena -> floorAppearance -> baseColorMap -> url -> change URL to some other file from the textures/ directory, either by the "Select..." dialog or by the text field
4. boom

Reduced case (may be caused by another bug):
1. File - open sample world - devices/led
2. In the object tree, left click on RectangleArena -> floorAppearance
3. Right click - convert to PROTO node is not shown (why?)
4. Right click on some other objects
5. Right click on floorAppearance - now "convert to PROTO" is shown
6. click on "convert to PROTO"
7. boom - webots crashes hard
8. if that bug is fixed, change the URL of the floorAppearance, that should also make it crash

**Expected behavior**
Does not crash

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: NVIDIA Quadro P1000 Mobile

**Additional context**
In my experience, editing/removing nodes is the top reason for crashes.

stefaniapedrazzi, 2020-07-08:
Thank you for reporting these crashes.
I'm already fixing these issues in #1868.

Just for note, the crash due to the convert to base action was also reported here (https://github.com/cyberbotics/webots/issues/1854).

stefaniapedrazzi, 2020-07-08:
The issues with the context menu will be fixed in  #1889

closed 2020-07-22 by stefaniapedrazzi