cyberbotics/webots #1883: Improve ROS installation instructions

stefaniapedrazzi, 2020-07-08:
I think it would be very helpful to re-add to the documentation the instructions to create the webots-ros package in the catkin workspace based on the current branch.
See:
- test script https://github.com/cyberbotics/webots/pull/1876/files#diff-3633cfd301418db46b354e5411c426e9 or
- simply restoring the old documentation https://www.cyberbotics.com/doc/guide/tutorial-7-using-ros?version=cyberbotics:R2019a#webots_ros-package-installation

One problem with the current installation instructions [Turotial 8](https://www.cyberbotics.com/doc/guide/tutorial-8-using-ros) is that they seems to work only for the latest version of Webots, but if a new `webots-ros` package is distributed that will break the backward compatibility, then the instructions in the older manual are useless and won't work.

DavidMansolino, 2020-07-08:
What about pointing to http://wiki.ros.org/webots_ros#From_Sources instead of duplicating this information?

I know that they refer to the files from the _webots_ros_ repo and not the _webots_ one but in theory, they should match, maybe we should remove completely these duplicated files from the _webots_ repo and use a submodule (I am always a bit hesitating to use sub-modules as we had a bad experience with it, but this looks like the 'perfect' use case)?

stefaniapedrazzi, 2020-07-08:
Yes, we could also add this.
But it won't explain how to use the data from the webots repo branch and we should make sure that each documentation describes the `webots_ros` package version used so that it is possible to get the sources of the correct one.

DavidMansolino, 2020-07-08:
> Webots, but if a new webots-ros package is distributed that will break the backward compatibility, then the instructions in the older manual are useless and won't work.

Just a side note about this, in theory, it is completely prohibited to break compatibility of a ROS package with an update (I saw many package updates been refused because of this), so in theory, once we release the package for a version of Webots, if we release a new version of Webots that is breaking backward compatibility we are not able to use this new version in the ROS package (which we are of course not respecting as this would cause many issues for us).

closed 2020-07-09 by stefaniapedrazzi