cyberbotics/webots #1886: Problem with touch sensor when importing nodes in run-time.

150012, 2020-07-08:
**Describe the Bug**
Problem with my touch sensor when importing nodes in run-time.

(See additional context on bottom first)
The touch sensor works fine when just moving the end-effector around. It returns 1.0 when it collides with other bounding objects, and 0.0 when it's not. But when I pick up an object (import it as a child of "Gripper"), it stops working. It will continue to return 0.0, even though the touch sensor overlaps other bounding objects. This is also the case when I manually add the object as a child of "Gripper" during run-time, so it seems like adding/importing objects in run-time might be part of the problem.
When I remove/delete the object (in run-time) in the child-field, everything works like before again.

(I can't upload the code or the world, because of an agreement in my job-contract).

**Expected behavior**
I expect the touch sensor to still notice when its bounding object overlaps with other bounding objects.

**Screenshots**
The "DEF default_box6 CardboardBox"-node is the object being imported in run-time.
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce MX250

**Additional context**
- I don't have any physics on, it's a pure kinematic simulation.
- The touch sensor is a child of an end-effector, along with the other children "Gripper" and "Camera". 
- The bounding object of the touch sensor covers the end-effector (thereby also the Gripper and Camera)).
- The "grab" function works by importing an object as a child of the "Gripper" (the object does not overlap with the bounding object of the touch sensor) in run-time, and then I delete/remove the original object.

150012, 2020-07-08:
When importing a solid without a bounding object instead, it works.

stefaniapedrazzi, 2020-08-12:
Are you sure that after importing the node the collision still occurs with the TouchSensor bounding object and not the new node bounding object?
Because in this case it would be correct that the collision is not detected.

I tried to reproduce this issue, but I couldn't.
If you cannot share your complete worlds, please a create and share a simple world where the same issue is reproducible so that we can further investigate this issue.

omichel, 2020-11-30:
I am closing this stalled issue as we cannot further investigate it for now. Feel free to re-open if needed.

closed 2020-11-30 by omichel