cyberbotics/webots #1903: shutdown of python controller does not close cleanly connection

oroulet, 2020-07-11:
When we run an extern python controller, sometimes (well most of the time) the communication is not correctly closed and restarting the controller leads to an error message "Failed to attach extern robot controller" [image]

The message "INFO '<extern>' controller existed successfully" does not appear on logging windows

We started discussing this on discord but I think it is easier here.

The controller is not my code so I am going through it but it seems there is no RobotController only a supervisor and some cameras. There is also no thread running.

to try to debug I made a simple script creating a supervisor and deleting it with del. it make the controller disconnect, so that works for a simple case

in our application I create the main object, close all threads then delete the supervisor but that does not disconnect. There is probably some dependencies preventing the C object to be deleted. so I tried to go through all objects and delete them by hand but it does not trigger the connection close....

Is there a way to find out what is keeping the connection open?

although generally I think it would suggest to allow to disconnect manually instead of using the delete methods of python objects. you could use the "with" statement for example. Implicit connection in python is often a nightmare...

DavidMansolino, 2020-07-13:
It would help a lot if you can share a controller  (as simple as possible) that reproduce this issue.

oroulet, 2020-07-13:
I will try ...

oroulet, 2020-07-29:
if you only ctrl-C once (and NOT twice) then the controller shuts down cleanly. Seems to be a second thread running that does not support interruption

oroulet, 2020-08-02:
Can be easily reproduced by starting webots then running following code:

```
import time
from threading import Thread

from controller import Supervisor

class T(Thread):
    def run(self):
        while not self.ev_stop:
            time.sleep(0.01)
            print(time.time())

s = Supervisor()

while s.step(5000) != -1:
    pass
```

then interupting with two times ctrl-C
If you restart the script one more time you get the Error on webots

Since this require two times ctrl-C it is very easy to avoid, thus not very high priority

DavidMansolino, 2020-08-03:
I can indeed reproduce it with your controller and double ctrl-C, however, with a simple ctrl-C it works fine. 
The second Ctrl+C causes the program to quit immediately, in that case the destructor of the `Supervisor` class is not called and the communication with Webots is closed without notifying Webots.

stefaniapedrazzi, 2021-06-17:
Should be fixed in #3189.

closed 2021-06-17 by stefaniapedrazzi