cyberbotics/webots #1904: MATLAB 2020a and WEBOTS 2020a

Abdullah-Mec, 2020-07-11:
**Describe the Bug**
When trying to run the following example "e-puck_matlab.wbt", the controller will fail to launch MATLAB

**Expected behavior**

MATLAB command window will launch and the following commands lines will appear:

Warning: Name is nonexistent or not a directory: C:\Users\username\AppData\Local\Programs\Webots\lib\matlab 
> In path (line 109)
  In addpath (line 86)
  In launcher (line 23) 
The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed, please install it from: https://fr.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-compiler
Waiting up to 5 seconds for C:\Users\username\AppData\Local\Temp\webots_matlab_lock to be deleted by another MATLAB instance.
Deleting C:\Users\username\AppData\Local\Temp\webots_matlab_lock...
Creating: C:\Users\username\AppData\Local\Temp\protofile_matlab_2020a_webots_R2020a_revision_1.m
Load Library failed.
Error using loadlibrary
Could not find file allincludes.h.

Error in loadlibrary

Error in launcher (line 74)
      loadlibrary( ...

»

**System**
 - Operating System: Windows 10, 64 bit
 - Graphics Card: NVIDIA GeForce RTX 1080ti 11 GB.]

Solution

Edit the following files as follows:

1. "launcher.m" under the following path "C:\Users\username\AppData\Local\Programs\Webots\lib\controller\matlab

Change the lines to:
Line 15:  cd('lib/controller/matlab');
Line 23: addpath([WEBOTS_HOME '/lib/controller/matlab']);
Line 132: cd([WEBOTS_HOME '/lib/controller/matlab']);

make sure to save the file after editing.

2. "allincludes.h" under the following path "C:\Users\username\AppData\Local\Programs\Webots\lib\controller\matlab

Change all the lines to the following (add an extra "../"):

#define WB_MATLAB_LOADLIBRARY

#include "../../../include/controller/c/webots/accelerometer.h"
#include "../../../include/controller/c/webots/brake.h"
#include "../../../include/controller/c/webots/camera.h"
#include "../../../include/controller/c/webots/compass.h"
#include "../../../include/controller/c/webots/connector.h"
#include "../../../include/controller/c/webots/console.h"
#include "../../../include/controller/c/webots/device.h"
#include "../../../include/controller/c/webots/differential_wheels.h"
#include "../../../include/controller/c/webots/display.h"
#include "../../../include/controller/c/webots/distance_sensor.h"
#include "../../../include/controller/c/webots/emitter.h"
#include "../../../include/controller/c/webots/gps.h"
#include "../../../include/controller/c/webots/gyro.h"
#include "../../../include/controller/c/webots/inertial_unit.h"
#include "../../../include/controller/c/webots/joystick.h"
#include "../../../include/controller/c/webots/keyboard.h"
#include "../../../include/controller/c/webots/led.h"
#include "../../../include/controller/c/webots/lidar.h"
#include "../../../include/controller/c/webots/light_sensor.h"
#include "../../../include/controller/c/webots/motor.h"
#include "../../../include/controller/c/webots/mouse.h"
#include "../../../include/controller/c/webots/pen.h"
#include "../../../include/controller/c/webots/position_sensor.h"
#include "../../../include/controller/c/webots/radar.h"
#include "../../../include/controller/c/webots/range_finder.h"
#include "../../../include/controller/c/webots/receiver.h"
#include "../../../include/controller/c/webots/robot.h"
#include "../../../include/controller/c/webots/skin.h"
#include "../../../include/controller/c/webots/speaker.h"
#include "../../../include/controller/c/webots/supervisor.h"
#include "../../../include/controller/c/webots/touch_sensor.h"
#include "../../../include/controller/c/webots/utils/motion.h"
#include "../../../include/controller/c/webots/utils/system.h"

make sure to save the file after editing.

it should be working now, but it will still report that
 "The MATLAB "MinGW-w64 C/C++ Compiler" addon is not installed"
Even though it's installed and added in MATLAB

Kind Regards
Abdullah

omichel, 2020-07-13:
Thank you for reporting this bug, including the fix.
However, I believe it was already fixed in Webots R2020a revision 2 and R2020b which you can download from here: https://github.com/cyberbotics/webots/releases (you will find them in the latest nightly builds).
Please let me know if that resolves the bug and we will be able to close this issue.

Abdullah-Mec, 2020-07-14:
Can confirm its solved using Webots 2020a rev2

omichel, 2020-07-14:
Thank you for the feedback! So I am considering this issue is resolved.

closed 2020-07-14 by omichel