cyberbotics/webots #1905: wb_supervisor_movie_is_ready cannot be used to detect that the previous movie has been written

PeterJCLaw, 2020-07-11:
**Describe the Bug**

The docs indicate that [`wb_supervisor_movie_is_ready`](https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_movie_is_ready) can be used to determine whether the previous video recording has completed and been written to disk. Unfortunately this doesn't seem to be the case.

**Steps to Reproduce**

Start webots with the following supervisor configured:
``` python
import time
from controller import Supervisor

supervisor.movieStartRecording(
    'that.mp4', width=1920, height=1080, quality=100,
    codec=0, acceleration=1, caption=False,
)
# our actual durations are ~150s, I'm assuming that this 3.2 sleep will be enough to show the issue
supervisor.step(3200)
supervisor.movieStopRecording()

while supervisor.movieIsReady():
    print('waiting')
    time.sleep(0.1)

movie_ok = not supervisor.movieFailed()
print('result:', movie_ok)

supervisor.simulationQuit(0)
```

This will print "result: True" almost immediately after the simulation pauses and will not print "waiting" at all. Webots will then quit but the recording will not be output.

**Expected behavior**

`supervisor.movieIsReady` should return `False` until the movie has been fully output.

**System**
 - Operating System: Ubuntu 18.04.4 also Arch

DavidMansolino, 2020-07-13:
Your controller has some issues, here is a corected version:
```
import time
from controller import Supervisor

supervisor = Supervisor()

supervisor.movieStartRecording(
    'that.mp4', width=1920, height=1080, quality=100,
    codec=0, acceleration=1, caption=False,
)
# our actual durations are ~150s, I'm assuming that this 3.2 sleep will be enough to show the issue
supervisor.step(3200)
supervisor.movieStopRecording()

while not supervisor.movieIsReady():
    print('waiting')
    time.sleep(0.1)

movie_ok = not supervisor.movieFailed()
print('result:', movie_ok)

supervisor.simulationQuit(0)
```

But I confirm there is an issue here as the state of `movieIsReady` is updated only when calling the step, which we don't want here (and also in some user use case of other projects).

DavidMansolino, 2020-07-13:
Fixed in  #1908

closed 2020-07-13 by DavidMansolino

PeterJCLaw, 2020-07-14:
D'oh, silly error on my part, Thanks for fixing so quickly!

DavidMansolino, 2020-07-14:
You're welcome.