cyberbotics/webots #1921: Camera tests no longer passing on  Ubuntu 18.04

stefaniapedrazzi, 2020-07-15:
At least since last week, some of the camera tests are no longer passing on my Ubuntu 18.08 development system.

```
...
FAILURE with camera_color: Wrong color (#0) at (5, 8), Received color = (121, 66, 56), Expected color = (123, 71, 58)
...
FAILURE with camera_color_spherical: Wrong color (#0) at (22, 12), Received color = (203, 203, 0), Expected color = (207, 207, 0)
...
FAILURE with camera_noise_mask: Wrong color at (5, 8), Received color = (121, 66, 56), Expected color = (123, 71, 58)
...
```

System info:
```
stefi@libeccio:~/webots$ ./webots --sysinfo
System: Linux 4.15.0-109-generic x86_64
Processor: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
Number of cores: 12
OpenGL vendor: NVIDIA Corporation
OpenGL renderer: GeForce RTX 2060/PCIe/SSE2
OpenGL version: 4.6.0 NVIDIA 440.100
```

I'm using the latest NVIDIA drivers available from the system.
I will try to download and install the latest version from the NVIDIA website.

omichel, 2020-07-15:
This is likely be caused by #1871. We should probably slightly increase the tolerance on the color tests.

stefaniapedrazzi, 2020-07-15:
Yes, it is just strange that they were working before and I didn't manually change the drivers version nor the hardware.

So this could be due to some system update or Webots changes.

DavidMansolino, 2020-07-15:
Just for info, the github action tests are running on Ubuntu 18.04 and are passing (test from yesterday evening: https://github.com/cyberbotics/webots/runs/870756625?check_suite_focus=true).
```
System              : Linux 5.3.0-1031-azure x86_64
Processor           : Intel(R) Xeon(R) CPU E5-2673 v3 @ 2.40GHz
Number of cores     : 2
OpenGL vendor       : VMware, Inc.
OpenGL renderer     : llvmpipe (LLVM 9.0, 256 bits)
OpenGL version      : 3.1 Mesa 19.2.8
```

stefaniapedrazzi, 2020-07-15:
Yes, for all of my recent PR, tests were failing locally but passing on the GitHub Actions.

stefaniapedrazzi, 2020-07-20:
I tested on the master branch and on R2020a tag and the tests are passing correctly.
So the failure doesn't seem to be due to some change in my system or graphics driver but rather to some changes in Webots develop branch.

stefaniapedrazzi, 2020-07-22:
I'm still looking for the commit that causes the difference in the tests.

stefaniapedrazzi, 2020-07-29:
Finally the issue was simply that the texture quality in my settings was set to Low.

closed 2020-07-29 by stefaniapedrazzi