cyberbotics/webots #1923: Modify internal PROTO nodes and fields from Supervisor API

stefaniapedrazzi, 2020-07-15:
Since #1331, it is now possible to modify internal fields of a PROTO node using the Supervisor API.
But this seems fundamentally wrong.
Should not be possible to modify only the PROTO parameters (the same as from the scene tree)?

In issue #1310, it was described the need to access internal PROTO data only in read-only mode but it doesn't seem that the read-only constraint has not been implemented.

DavidMansolino, 2020-07-15:
This should not be the case, we should only be allowed to read inter fields but not to change them:
https://github.com/cyberbotics/webots/blob/develop/src/lib/Controller/api/supervisor.c#L990
Did you found a case where the `is_proto_internal` flag is not correct?

stefaniapedrazzi, 2020-07-15:
I have a sample simulation where I can change internal fields.
I will check the `is_proto_internal` flag.

stefaniapedrazzi, 2020-07-15:
I confirm the `is_proto_internal` flag is not set.
This is because the field is not an internal field of a PROTO node, but a field of a base node defined in a PROTO body.
The current check loops only to the parent node field and doesn't take into account if the ancestor nodes.

stefaniapedrazzi, 2020-07-15:
Here is a sample node structure where you can change internal fields.

```
PROTO Test [
]
{
Robot {
  children [
    DEF IINTERNAL Solid {
      translation 0 0 0
    }
  ]
}
}
```

controller code:
```
WbNodeRef internal = wb_supervisor_node_get_from_proto_def("INTERNAL");
WbFieldRef internal_field = wb_supervisor_node_get_field(internal, "translation");
const double t[3] = { 0, 0.5, 0 };
wb_supervisor_field_set_sf_vec3f(internal_field, t);
```

DavidMansolino, 2020-07-15:
Ok, then this is probably a bug :disappointed:

stefaniapedrazzi, 2020-07-16:
I'm fixing it in #1925.

But checking the new `wb_supervisor_node_get_from_proto_def` and `wb_supervisor_node_get_proto_field` functions, I'm wondering if they should not really work only for PROTO nodes (now they work for every node) and if we should not pass the parent PROTO node reference in order to limit the scope of the search because currently the first DEF found is returned independently if it is in a PROTO body or not.

DavidMansolino, 2020-07-16:
Why not, but in that case, we should at least provide a function to know if a node is a PROTO or not.

omichel, 2020-07-16:
Yes, I also believe these functions should work for PROTO nodes onles.
Also `wb_supervisor_node_get_from_proto_def` should take the proto node as a parameter.

stefaniapedrazzi, 2020-07-16:
Ok. So if we agree I will modify the functions (so that it would also be maybe possible to improve the fix in #1925) and add a new function that returns if a node is a PROTO:
```
wb_supervisor_node_is_proto(WbNodeRef node)
```

PR #1925 (2020-07-21): Fix internal proto field flag in supervisor.c
Fix #1923: improve the checks to detect if a field is an internal PROTO field.
This flag is needed in the libController to prevent modifying internal PROTO fields.

Previously only direct fields of the PROTO node were identified.
This PR extends the detection also to field of nodes defined in the PROTO body.
closed 2020-07-21 by stefaniapedrazzi