cyberbotics/webots #1936: Reason image not found abort trap 6

AlbusSong, 2020-07-19:
I just updated the latest codes on branch **master**. And without any trouble, I built the codes.
But when I tried to launch the app, it threw me such an error.

[image]

So anyone has some ideas?

DavidMansolino, 2020-07-20:
Are you sure you are compiling from master?

I just checked and on master libassimp has no dependency on libIrrXML:
```shell
David@Cyberboticss-Mac-mini webots % otool -L lib/webots/libassimp.dylib 
lib/webots/libassimp.dylib:
	@rpath/lib/webots/libassimp.dylib (compatibility version 4.0.0, current version 4.1.0)
	/usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
```

Or maybe you still have some artefacts from develop.
Can you try recompiling after a git clean?
```shell
git clean -fdfx
make release -j12
```

AlbusSong, 2020-07-20:
I am sure. 
Well, to make it 100% sure, now I will delete all things and reclone the codes from Github.

DavidMansolino, 2020-07-20:
Ok, thank you, but note that the default clone will clone the develop branch, to clone the master use the following command:
```shell
git clone --branch master https://github.com/cyberbotics/webots.git
```

AlbusSong, 2020-07-20:
Thank you. But in fact I know how to switch branches.

Just wonder, which command shall I use?
`git clone --branch master --recurse-submodules -j8 https://github.com/cyberbotics/webots.git`
or 
`git clone --branch master https://github.com/cyberbotics/webots.git`

omichel, 2020-07-20:
`git clone --branch master --recurse-submodules -j8 https://github.com/cyberbotics/webots.git` is the way to go.
But if you already cloned the repo, you may simply use `git branch master` instead, that's way faster and easier.

AlbusSong, 2020-07-20:
Luckily, it's fine now!
Sorry for the bother

closed 2020-07-20 by AlbusSong