cyberbotics/webots #1942: Applying a torque on an object connected to another via a connector won't work

Acwok, 2020-07-20:
**Describe the Bug**
I am applying a torque on a robot using a supervisor controller after the first two seconds of the simulation.  
The physicsDisbleTime parameter is set to 1s.

The torque is not applied when:
- the robot is connected to another object through a connector node
- the torque is applied after the `physicsDisableTime` parameter of the WorldInfo

We think that the `physicsDisableTime` disable the physics of the green cube (which is connected to the red cube through a connector), and this prevent the red cube from rotating.
If we set the connector isLocked fields to FALSE, the red cube is moving correctly.
We also noticed that updating the physics by slightly moving manually one of the cube, the red cube starts rotating.

See this video: https://youtu.be/DQWD7DMpvlI
Here is the world with the controller: [physicsDisableTimeTest.zip](https://github.com/cyberbotics/webots/files/4946845/physicsDisableTimeTest.zip)

**Steps to Reproduce**
1. open the world
2. make sure the red cube has the physicsDisableTimeTest controller attached
3. Run the simulation

**Expected behavior**
The red cube should start rotating after 2s of simulation and drag the green cube with him.

**System**
- Ubuntu 18.04 in a docker with GPU acceleration
- Webots build from sources commit 65e5a49

DavidMansolino, 2020-07-20:
You are completely right, we will make sure to remove the sleep flag to the node to which a force is added.

Acwok, 2020-07-20:
> we will make sure to remove the sleep flag to the node to which a force is added.

On second thought, I think it won't work.  
In the video you can see that when the green cube is not locked to the red cube, the torque is well applied.  
The problem rather comes from the green cube whose physics is also disabled.

So, the sleep flag should be removed for all the nodes connected to the node to which a force is added.

closed 2020-07-21 by DavidMansolino