cyberbotics/webots #1944: Pop Alert Window Buttons Can't Respond

AlbusSong, 2020-07-20:
These buttons are unable to respond to user's click actions.

[image]

DavidMansolino, 2020-07-20:
Do you experience this only with these buttons or all of them?

DavidMansolino, 2020-07-20:
I can indeed reproduce this with all the pop-up windows.

omichel, 2020-07-20:
It seem to be a [known bug](https://bugreports.qt.io/browse/QTBUG-84879).

AlbusSong, 2020-07-20:
Oh, would have to wait a bit long. 
So what should I do to avoid this bug?
Or you will do something to fix it before August 20th?

omichel, 2020-07-20:
Yes, we will fix it in the next couple of days (probably with Qt 5.14.2), so that you can use a nightly build that corrects the problem.

DavidMansolino, 2020-07-21:
Here is the update PR #1736, I propose to simply rollback this PR while waiting for Qt 5.15.1

AlbusSong, 2020-07-21:
OK, so fast.
So what I shall do now is just to wait for the merging of the **pr** on **master** branch, right?

omichel, 2020-07-21:
On the **develop** branch, yes. Then, you will be able to download the next R2020b nightly build which will include the patch.

closed 2020-07-21 by DavidMansolino