cyberbotics/webots #1945: Build Broken on macOS

DavidMansolino, 2020-07-20:
**Describe the Bug**
Currently the main build is broken on macOS in develop:
```shell
# updating  build/release/abstract_camera.d
build/release/abstract_camera.d:3: *** missing separator.  Stop.
make: *** [webots_target] Error 2
```

DavidMansolino, 2020-07-21:
As expected, it is the 7eea8981e3e9fc3a2b52bac65fe74189db20dc66 commit that is responsible of this.

DavidMansolino, 2020-07-21:
Before 7eea8981e3e9fc3a2b52bac65fe74189db20dc66
```
build/release/abstract_camera.o: build/release/abstract_camera.d

abstract_camera.o: api/abstract_camera.c api/abstract_camera.h \
  ../../../include/controller/c/webots/types.h api/device_private.h \
  ../../../include/controller/c/webots/nodes.h api/request.h \
  api/messages.h api/robot_private.h \
  ../../../include/controller/c/webots/supervisor.h
```
After:
```
build/release/abstract_camera.o: build/release/abstract_camera.d

-n build/release/
abstract_camera.o: api/abstract_camera.c api/abstract_camera.h \
  ../../../include/controller/c/webots/types.h api/device_private.h \
  ../../../include/controller/c/webots/nodes.h api/request.h \
  api/messages.h api/robot_private.h \
  ../../../include/controller/c/webots/supervisor.h
```

Unexpectedly, on Linux on develop it seems 7eea8981e3e9fc3a2b52bac65fe74189db20dc66 has not effect on the *.d file.

omichel, 2020-07-21:
It looks like `echo -n` is not well interpreted...

DavidMansolino, 2020-07-21:
https://pubs.opengroup.org/onlinepubs/9699919799/utilities/echo.html:
> If the first operand is -n, or if any of the operands contain a <backslash> character, the results are implementation-defined.

That probably explains the problem.

omichel, 2020-07-21:
Yes, we should probably go for the `\c` sequence instead.

DavidMansolino, 2020-07-21:
A simple solution would be to write the two entries on one line:
```diff
 $(OBJDIR)/%.d:%.c
        @echo "# updating " $@
-       @echo $(basename $(OBJDIR)/$(notdir $@)).o: $(basename $@).d > $(OBJDIR)/$(notdir $@)
+       @echo $(basename $(OBJDIR)/$(notdir $@)).o: $(basename $@).d $(OBJDIR)/$(notdir $@)/ > $(OBJDIR)/$(notdir $@)
        @echo >> $(OBJDIR)/$(notdir $@)
-       @echo -n $(OBJDIR)/ >> $(OBJDIR)/$(notdir $@)
        @$(CC) $(CFLAGS) -MM -w $(INCLUDE) $< >> $(OBJDIR)/$(notdir $@)
 ```

DavidMansolino, 2020-07-21:
> Yes, we should probably go for the `\c` sequence instead.

yes that should work too.

closed 2020-07-21 by DavidMansolino