cyberbotics/webots #1958: could not start 'make'&Unsupported image format.

yudodisss, 2020-07-22:
**Describe the Bug**
could not start 'make'&Unsupported image format.

**Steps to Reproduce**
1. install webots
2. launch webots,everything is ok.
3. close webots
4. launch webots,there is someting wrong.

**Expected behavior**
there are many warnings appear on the console.for example:"WARNING: E-puck (PROTO) > DEF EPUCK_TURRET Transform > Shape > PBRAppearance > ImageTexture: Cannot load texture 'D:/Webots/projects/robots/gctronic/e-puck/protos/textures/e-puck1_turret_base_color.jpg': Unsupported image format."
And then click "build the current project" button,"make    Installation problem: could not start 'make'."

**System**
 -  Windows 10
 - Graphics Card: lntel(R)HD Graphics 620

omichel, 2020-07-22:
Thank you for this bug report. However, I believe this is a known bug which was already fixed in our development environment. Can you try a [nightly build](https://github.com/cyberbotics/webots/releases) of Webots and let us know if it fixes the problem? I would recommend testing [this one](https://github.com/cyberbotics/webots/releases/download/nightly_20_7_2020/webots-R2020b_setup.exe).

yudodisss, 2020-07-23:
Thank for your help.However I have trouble in downloading exe file,so I try to download source files and compile by myself.I follow [these steps](https://github.com/cyberbotics/webots/wiki/Windows-installation)，（didn't Install Optional Dependencies [ optional ]）.However when I type "make -j12"，errors appeared as below.How can I solve the problem?It's my first time to compile such a big program without IDE.Could you help me,please?

Thinkpad@۽▒ MSYS /d/github/webots
make -j12

Python not installed, skipping documentation

* dependencies *

compiling lua
make[4]: gcc: No such file or directory
make[4]: *** [<builtin>: lua.o] Error 127
make[3]: *** [Makefile:112: mingw] Error 2
make[2]: *** [Makefile:55: mingw] Error 2
make[1]: *** [Makefile.windows:116: D:/github/webots/dependencies/lua-5.2.3/src/lua52.dll] Error 2
make: *** [Makefile:128: webots_dependencies] Error 2

omichel, 2020-07-23:
It seems you cannot compile lua...
If gcc properly installed? Can you type `gcc --version`?
If so, did the installation process downloaded successfully the lua package named `lua-5.2.3.tar.gz` in your `webots/dependencies` folder?
Can you try to:
```bash
cd webots/dependencies
make -f Makefile.windows cleanse
make -f Makefile.windows
```
This will clean the dependencies folder and try again to build it, displaying possible errors.
Here is my output for this:
```bash
$ make -f Makefile.windows
# copying clang-format configuration file
# downloading openvr-1.0.7.zip
# uncompressing openvr-1.0.7.zip
# downloading lua-5.2.3.tar.gz
# uncompressing lua-5.2.3.tar.gz
# compiling lua
# copying lua dll
# downloading libOIS.zip
# uncompressing libOIS.zip
# downloading libpico.zip
# uncompressing libpico.zip
# downloading lua-gd-windows.zip
# uncompressing lua-gd-windows.zip
```

yudodisss, 2020-07-23:
Thank you very much!Though I still can't compile the project,I download the exe file in some way.I install and try the new webots,it performs well! 
I'll learn more about compilation in the future.
Thanks for your help!

closed 2020-07-24 by omichel