cyberbotics/webots #1963: Pioneer3Gripper falls apart when moving with a UR-5 robot

culurciello, 2020-07-22:
Hi, I am using a Pioneer3Gripper with a UR-5 robot. As you can see from the image below, the gripper falls apart when the robot is moving moving. What would be the cause of this? If anyone can point me in the right direction I will try to fix the gripper PROTO file

[image]

Thanks a lot~
E

DavidMansolino, 2020-07-23:
Can you try with the `P-Grip` one (https://github.com/cyberbotics/webots/blob/develop/projects/robots/fp_robotics/p-rob3/protos/P-Grip.proto) to see if the problem is specific to this gripper?

culurciello, 2020-07-23:
Thanks @DavidMansolino this gripper does not have that problem == it stay in one piece during arm motion! Will test more now.

DavidMansolino, 2020-07-24:
Ok, good. 
About the original gripper, which gripper exactly are you using?
The one on the picture doesn't look like the `Khepera3_Gripper` one.

DavidMansolino, 2020-07-24:
Ok, good.
About the original gripper, to rotate the gripper, did you add an intermediate Transform or Solid node?

omichel, 2020-11-30:
I am closing this stalled issue. Feel free to re-open if needed.

closed 2020-11-30 by omichel