cyberbotics/webots #1966: Worlds throwing errors

stefaniapedrazzi, 2020-07-22:
I started to test all the distributed worlds (same order as in the Open Sample World dialog). Here are the ones that throw errors:
* inverse_kinematics.wbt:
```
INFO: inverse_kinematics: Starting controller: python3 -u inverse_kinematics.py
INFO: valve_turner: Starting controller: python3 -u valve_turner.py 15.7079632679
INFO: valve_turner: Starting controller: python3 -u valve_turner.py 15.7079632679
Draw a circle on the paper sheet...
Traceback (most recent call last):
  File "inverse_kinematics.py", line 96, in <module>
    ikResults = armChain.inverse_kinematics([x, y, z])
  File "/usr/local/lib/python3.6/dist-packages/ikpy/chain.py", line 102, in inverse_kinematics
    raise ValueError("Your target must be a 4x4 transformation matrix")
ValueError: Your target must be a 4x4 transformation matrix
WARNING: 'inverse_kinematics' controller exited with status: 1.
```

* e-puck_kinematic.wbt: this warning is displayed at each controller loop step
```
WARNING: E-puck (PROTO) > DEF EPUCK_ACCELEROMETER Accelerometer: Parent of Accelerometer node has no physics: measurements may be wrong.
```

[to be completed]

DavidMansolino, 2020-07-22:
About ``inverse_kinematics``, which version of the `ikpy` module are you using?
You are probably not using the latest one (see #1782)

stefaniapedrazzi, 2020-07-22:
Yes, probably I don't have the latest version. But it should not only be checked if ikpy is installed but also that the version is the expected one.

stefaniapedrazzi, 2020-07-23:
I just completed opening and running all the worlds.

My ikpy version is 2.3.2.

stefaniapedrazzi, 2020-07-27:
I will start fixing the problem with the internal proto flag for Supervisor *self* node.

stefaniapedrazzi, 2020-07-27:
`motor.wbt` seems quite unstable.
I tried to change the masses of the solids and I could reduce the number of ODE errors printed, but as soon as the top face of the cylinder touches the face of the box and error remains.

PR #2025 (2020-07-28): Fixed stability of motor sample
Fixes #1966 (last item).
The instability was due to a closed mechanical loop.
Making the box lie on a the static world (instead of the dynamics object hitting it with the needle) broke this loop and made the simulation rock stable.
closed 2020-07-28 by omichel