cyberbotics/webots #1971: The salamander falls below the ground

omichel, 2020-07-23:
On Windows, the salamander robot in the `animated_skin.wbt` demo of the guided tour falls below the ground and disappear.

stefaniapedrazzi, 2020-07-23:
On Ubuntu 18.04 it works correctly.

DavidMansolino, 2020-07-23:
On macOS it crashes at load:
```
Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libode.dylib                  	0x0000000104bcf3ac dxTriMeshData::Build(void const*, int, int, void const*, int, int, void const*, bool) + 1148 (collision_trimesh_opcode.cpp:144)
1   libode.dylib                  	0x0000000104bd0c44 dGeomTriMeshDataBuildDouble1 + 132 (collision_trimesh_opcode.cpp:459)
2   libode.dylib                  	0x0000000104bd0caa dGeomTriMeshDataBuildDouble + 90 (collision_trimesh_opcode.cpp:467)
3   com.cyberbotics.webots        	0x0000000102e345bd WbTriangleMeshGeometry::setOdeTrimeshData() + 253 (WbTriangleMeshGeometry.cpp:307)
4   com.cyberbotics.webots        	0x0000000102e3443a WbTriangleMeshGeometry::createOdeGeom(dxSpace*) + 170 (WbTriangleMeshGeometry.cpp:289)
5   com.cyberbotics.webots        	0x0000000102c7944e WbMatter::createOdeGeomFromGeometry(dxSpace*, WbGeometry*, bool) + 78 (WbMatter.cpp:275)
6   com.cyberbotics.webots        	0x0000000102c7be7a WbMatter::createOdeGeomFromNode(dxSpace*, WbBaseNode*) + 1418 (WbMatter.cpp:490)
7   com.cyberbotics.webots        	0x0000000102c7b8e8 WbMatter::createOdeGeomFromBoundingObject(dxSpace*) + 56 (WbMatter.cpp:442)
8   com.cyberbotics.webots        	0x0000000102dab133 WbSolid::createOdeGeoms() + 195 (WbSolid.cpp:899)
9   com.cyberbotics.webots        	0x0000000102da9762 WbSolid::setupSolidMerger() + 194 (WbSolid.cpp:786)
10  com.cyberbotics.webots        	0x0000000102da9ccc WbSolid::setupSolidMergers() + 28 (WbSolid.cpp:806)
11  com.cyberbotics.webots        	0x0000000102daa04f WbSolid::createOdeObjects() + 111 (WbSolid.cpp:831)
12  com.cyberbotics.webots        	0x0000000102bf6954 WbGroup::createOdeObjects() + 100 (WbGroup.cpp:148)
13  com.cyberbotics.webots        	0x0000000102bf6954 WbGroup::createOdeObjects() + 100 (WbGroup.cpp:148)
14  com.cyberbotics.webots        	0x0000000102b3b5f7 WbBaseNode::finalize() + 519
15  com.cyberbotics.webots        	0x0000000102d8968a WbSimulationWorld::WbSimulationWorld(WbProtoList*, WbTokenizer*) + 1354 (WbSimulationWorld.cpp:84)
```

lukicdarkoo, 2020-07-23:
Same for the snap package and Ubuntu 20.04

closed 2020-07-24 by omichel