cyberbotics/webots #1975: Sojourner move through rocks

DavidMansolino, 2020-07-23:
**Describe the Bug**
The Sojourner moves through the rock as if it was not there:
[image]

stefaniapedrazzi, 2020-07-23:
On Ubuntu 18.04 it works fine.

omichel, 2020-07-23:
On Windows, it is very strange, I can reproduce this bug, but not systematically. I am trying to identify a reproducible procedure.

omichel, 2020-07-23:
I could not find a way to reproduce the problem systematically, but with the following procedure it is likely (about 50%) to show the buggy behavior:
1. Open webots with projects/langauges/cpp/worlds/example.wbt
2. Go to the File menu, Open Sample World and select the sojourner.wbt world by double-clicking on it.
3. In about 50% of the cases, the robot passes through the first stone.
4. Reset simulation can show the bug again systematically.
5. Revert simulation fixes the problem systematically.

stefaniapedrazzi, 2020-07-23:
With this procedure it is reproducible on Ubuntu too.

omichel, 2020-07-23:
With this simple world file, I can reproduce a similar bug on Windows.
In about one run out of 4 (you need to perform several reverts to see it), the ball is passing through the rock.
```
#VRML_SIM R2020b utf8
WorldInfo {
  coordinateSystem "NUE"
}
Viewpoint {
  orientation -0.8948797260489955 -0.4263143137077413 -0.13208475246739285 0.6620073677969002
  position 0.1867745168165805 0.49890843050016187 0.6135971533594632
}
Background {
  skyColor [
    0.15 0.45 1
  ]
}
DirectionalLight {
  ambientIntensity 1
  direction 0 -1 0
  castShadows TRUE
}
Rock10cm {
  translation 0.387314 -0.116121 -0.177911
  name "rock 10 cm(2)"
  color 0.9 0.4 0.3
}
Ball {
  translation 0.39 0.09716 -0.166909
}
```

omichel, 2020-07-24:
It seems to be a regression as I cannot reproduce the bug on the master branch.

PR #1994 (2020-07-24): Fixed random triangle mesh collision detection bug
Fixes #1975.

The triangle mesh passed to ODE trimesh constructor was wrong. The number of vertices was three times too much causing ODE to read uninitialized garbage data beyond the actual vertices data, sometimes including nan values and other oddities. This was causing the AABB of the resulting mesh to be wrong, and the ODE "near" callback was never called, thus missing the collision detection.
closed 2020-07-24 by omichel