cyberbotics/webots #1976: Camera auto-focus not working on macOS

DavidMansolino, 2020-07-23:
**Describe the Bug**
The focus is not done on macOS:
[image]

omichel, 2020-07-23:
On Windows, it is working properly.

DavidMansolino, 2020-07-23:
On Ubuntu too.

DavidMansolino, 2020-07-23:
The focus field is correctly changing from ~0.1 to ~0.6 as on Ubutnu, but there is no visible impact on the camera image.

DavidMansolino, 2020-07-24:
I checked on R2020a-rev1 and this is not a regression.
It actually seems that the focus is somehow working, but the result is not as nice as on the other platforms, the non-blur part is way less visible.

DavidMansolino, 2020-07-29:
Since this PR is not a regression and is probably due to the GPU and not a bug I will postpone it so that we can keep going with the release.