cyberbotics/webots #1980: Non-Working AINSI Escape code.

DavidMansolino, 2020-07-23:
**Describe the Bug**
In the Python example code when pressing keyboard keys (e.g. S and F), the controllers are displaying raw AINSI code:
```
Commands:
 I for displaying the commands
 A for avoid obstacles
 F for move forward
 S for stop
 T for turn
 R for positioning ROBOT1 at (0.1,0.3)
 G for knowing the (x,z) position of ROBOT1
Please, stop
I should \u001b[31mstop\u001b[0m!
I should \u001b[31mstop\u001b[0m!
I should \u001b[31mstop\u001b[0m!
Please, move forward
I should \u001b[31mmove forward\u001b[0m!
I should \u001b[31mmove forward\u001b[0m!
I should \u001b[31mmove forward\u001b[0m!
```

omichel, 2020-07-23:
On Windows, that works correctly.

DavidMansolino, 2020-07-23:
I can reproduce it on Ubuntu too.

DavidMansolino, 2020-07-24:
Actually it works fine with python3 but not with python2.

DavidMansolino, 2020-07-24:
Here is the message received by Webots:
from 2.7: `I should \\u001b[31mstop\\u001b[0m!\n`
from 3.7: `I should \u001B[31mstop\u001B[0m!\n`

omichel, 2020-07-24:
I believe a solution could be to replace `\u001b` with `\033` in the python `controller.i`.

closed 2020-07-24 by DavidMansolino