cyberbotics/webots #1985: Multiple issues with LiDAR

lukicdarkoo, 2020-07-23:
**Describe the Bug**
HokuyoUtm30lx sensor shows wrong readings.

**Steps to Reproduce**
1. Run `webots /snap/webots/x1/usr/share/webots/projects/samples/devices/worlds/hokuyo.wbt`
2. Check the screenshots for incorrect behaviours

**Screenshots**
`Show Lidar Rays Paths` and `Show Lidar Point Cloud`
[image]
Default, no changes in the world, laser rays are not visible even though `Show Lidar Rays Paths` is on. There are some lines, but no rays (although I am not sure what I am supposed to see). Also, no point cloud when `Show Lidar Point Cloud`. According to the documentation, I think I should see rays and points: https://cyberbotics.com/doc/guide/the-user-interface

`minRange`
[image]
If we assume that the blue circle is `minRange` something is not correct according to the display in the up-left corner.

`noise` 
[image]
Noise is set to 0.005 meaning that the standard deviation is 15cm. Distances shown on the display, that represent the distance to the green box, are sometimes huge.
This is most likely related to #1784 

**System**
 - Operating System: Linux Ubuntu 20.04
-  Graphics Card: Intel UHD Graphics 620

DavidMansolino, 2020-07-23:
> Default, no changes in the world, laser rays are not visible even though Show Lidar Rays Paths is on. There are some lines, but no rays (although I am not sure what I am supposed to see). Also, no [...]

You don't see the point-cloud even if enabled in the optional rendering because the point-cloud mode the lidar is not enabled. To test point-cloud optional rendering, check the devices/lidar simulation.

lukicdarkoo, 2020-07-23:
@DavidMansolino Sorry, this is my bad, I will edit the issue.

PR #2509 (2020-12-09): Make Lidar return INF if depth is out-of-range
**Description**
Previously, the LiDAR would return MAX_RANGE if the depth is out-of-range. The behavior is hard to handle in robotic applications as we don't know whether the depth is equal to MAX_RANGE (consider the noise as well) or the depth is out-of-range.

**Analysis of the similar systems**

LDS-01
It seems it doesn't return measurements at all indices
https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver/blob/master/src/hlds_laser_publisher.cpp#L110
This probably depends on the measurement quality or range. I have this LiDAR in Lausanne so th [... cut]
closed 2020-12-10 by lukicdarkoo