cyberbotics/webots #1995: New solid falls down from the table.

xxchenchen, 2020-07-26:
**Describe the Bug**
My new solid falls down from the table.
[ Moreover I cant hold them by robot hand. It is not solid. Hand moves into the box.]

**Steps to Reproduce**
1. Go to 'add solid.' 
2. Move into the children 'your solids.' (Solid from new 3d models)
3. Select bounding object is a box
4- Select physic physic
5. error: My new solid falls down from the table. Moreover it is invisible not solid. Robot does not grasp it.

**Expected behavior**
It is expected that it is on the table not under. Moreover robot grasp it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
Windows and Ubunutu
webots 2020a and 2002b
Nvdia

[image]
[image]
[image]
[image]

DavidMansolino, 2020-07-27:
From the error message in your 3rd screenshot, it seems the 'kutu' Solid node misses a bounding object, which seems to be contradictory with what is visible in the last screenshot, is it the same simulation?

xxchenchen, 2020-07-27:
That is same simulation. Thanks

xxchenchen, 2020-07-27:
[image]
That is clean world. That inlcludes just one kutu solid.

xxchenchen, 2020-07-27:
[image]

DavidMansolino, 2020-07-27:
Can you please share the world file?

xxchenchen, 2020-07-27:
I sent it.

DavidMansolino, 2020-07-27:
There are several problem in your world file:
  - If you define the mass you have to set the density to -1.
  - Your bounding object is clearly too WAY too small 0.001 * 0.001 => 0.001 mm for the side of the box.

closed 2020-07-27 by DavidMansolino

xxchenchen, 2020-07-27:
[image]

xxchenchen, 2020-07-27:
density=-1 mass=0.5 boudnary size 0.15 0.15 0.04

I obtained errror. Why?

xxchenchen, 2020-07-27:
[image]
There is no error. However, the box falls down.