cyberbotics/webots #1996: supervisor.getSelf().remove() crashes Webots

stefaniapedrazzi, 2020-07-27:
A user reported that removing the self Supervisor node from a controller makes Webots crashing.

Not sure if it is feasible to support this.
If not, we should print an error and forbid it.

DavidMansolino, 2020-07-27:
I can indeed reproduce the bug by simply adding this line in the supervisor sample:
```C
  wb_supervisor_node_remove(wb_supervisor_node_get_self());
```

DavidMansolino, 2020-07-27:
Here is the stack:
```
#0  0x00007f6ec01102e2 in QQmlData::isSignalConnected(QAbstractDeclarativeData*, QObject const*, int) () from /home/david/webots/lib/webots/libQt5Qml.so.5
#1  0x00007f6ed0821b52 in void doActivate<false>(QObject*, int, void**) ()
   from /home/david/webots/lib/webots/libQt5Core.so.5
#2  0x0000000000b7c82f in WbSupervisorUtilities::worldModified (this=0x86d1700)
    at build/debug/WbSupervisorUtilities.moc.cpp:196
#3  0x0000000000a29e6f in WbSupervisorUtilities::handleMessage (
    this=0x86d1700, stream=...) at nodes/utils/WbSupervisorUtilities.cpp:1185
#4  0x00000000009acfa6 in WbRobot::handleMessage (this=0x86ca790, stream=...)
    at nodes/WbRobot.cpp:1031
#5  0x00000000009abc6e in WbRobot::dispatchMessage (this=0x86ca790, stream=...)
    at nodes/WbRobot.cpp:801
#6  0x00000000008513db in WbController::readRequest (this=0x171364f0)
    at control/WbController.cpp:1053
```

stefaniapedrazzi, 2020-07-27:
On my system I got a different stack, but the point should be the same: a node pointer is used after deleting it

closed 2020-07-28 by DavidMansolino