cyberbotics/webots #2022: Problem with Touch Sensor and PROTO's (and/or Transform node).

150012, 2020-07-28:
**Describe the Bug**
I'm using a Touch Sensor (bumper) to detect when its bounding object overlap other bounding objects. It works well before I add my own PROTO. The problem seems to appear when I add a Transform inside a Group node (for the PROTO's bounding object) - then it suddenly stops detecting other bounding objects, that it previously did detect.

**Steps to Reproduce**
1. Create a touch sensor (bumper) that detect when its bounding object overlap other bounding objects.
2. Import Webots' CardboardBox (https://cyberbotics.com/doc/guide/object-factory#cardboardbox) to verify that the touch 
    sensor can detect the boxes.
3. Create a PROTO as shown in the pictures below.
    First use only the "USE BEAM1", to see that everything works properly. Then swap "USE BEAM1" with the Transform as shown 
    in the picture.
4. See error.

**Expected behavior**
Expected the touch sensor to detect bounding objects as usual, with or without the Transform-node.

**Screenshots**
[image]
[image]

**System**
 - Operating System: Windows 10.
 - Graphics Card: NVIDIA GeForce MX250

**Additional context**
My touch sensor is located at [-1.12, 0.299998, 0] with the rotation [0, 1, 0, 1.57]. The size of the touch sensor's bounding object is [0.115, 0.134, 0.2].

150012, 2020-07-28:
If I remove the "translation 0.5185 1.5025 0" from the code, the touch sensor works again. It also works if I manually translate the bounding object in the world.

150012, 2020-07-28:
Update: If I start the simulation (after reloading) with the Touch Sensor being above every other bounding object (in y-axis), everything works just fine. Seems like the problem is somewhere around "Transform" / "translation" in PROTOs being higher up than the Touch Sensor..

omichel, 2020-07-28:
Could you please upload a simple project in a ZIP file that includes a simple world file, your simple proto files (two versions) and a simple controller program?

lukicdarkoo, 2020-11-30:
@150012, which Webots version did you use? This might be related to #2455 and #2420.

omichel, 2020-12-08:
@150012: I tried, but was unable to reproduce the problem. Everything works as expect for me in Webots R2020b-rev1. We really need that you upload a ZIP file of your project so that we can reproduce the problem and investigate it.

sigmunau, 2020-12-16:
The issue happens when the following code is run:
supervisor.getRoot().getField("children").importMFNodeFromString(-1, 'DEF AGV_Pallet1 currence_AGV_pallet { }')

with the following proto:
```
#VRML_SIM R2020b utf8
# license: Copyright Cyberbotics Ltd. Licensed for use only with Webots.
# license url: https://cyberbotics.com/webots_assets_license
# tags: static
# Simple and configurable wooden pallet.

PROTO currence_AGV_pallet [
  field SFVec3f    translation    -1.48 -100 0
  field SFRotation rotation       0 1 0 1.5708
  field MFNode     children       []
  field SFString   name           "wooden pallet"
  field SFVec3f    size           0.8 0.144 1.2    # Defines the size of the pallet.
  field SFInt32    lathNumber     4                # Defines the number of lath of the pallet.
  field SFFloat    lathWidth      0.145            # Defines the width of one lath.
  field SFBool     boundingObject TRUE             # Defines whether the bounding object should be enabled.
]

{
%{
  local size = fields.size.value

  lathNumber = fields.lathNumber.value
  lathWidth = fields.lathWidth.value

}%
Solid {
  translation IS translation
  rotation IS rotation
  children [

      DEF stack_frame Transform {
          translation 0 0.144 0
          rotation 0 1 0 -1.5708

          children IS children
      }
    # Center cube
    %{ for i=-1,1 do }%
      Transform {
        translation %{= size.x * 0 * 0.45 }% %{= 0.061 }% %{= size.z * i * 0.44 }%
        children [
          Shape {
            appearance RoughPine {
              textureTransform TextureTransform {
                scale %{= 2 * size.x * 0.1 }% %{= 2 * size.z * 0.1 }%
              }
            }
            geometry Box {
              size %{= size.x * 0.181 }% %{= size.y * 0.542 }% %{= size.z * 0.121}%
            }
          }
        ]
      }
    %{ end }%
    # Side cubes
    %{ for i=-1,1 do }%
      Transform {
        translation %{= -0.35 }% %{= 0.061 }% %{= size.z * i * 0.44 }%
        children [
          Shape {
            appearance RoughPine {
              textureTransform TextureTransform {
                scale %{= 2 * size.x * 0.1 }% %{= 2 * size.z * 0.1 }%
              }
            }
[... 112 lines cut ...]
  ]
  name IS name
  model "wooden pallet"
  %{ if fields.boundingObject.value then }%
  boundingObject Group {
    children [
      Transform {
        translation 0 %{= size.y / 2 }% 0
        children [
          Box {
            size IS size
          }
        ]
      }
    ]
  }
  %{ end }%

}
}

```
after this the touch sensor stops working (with any object, not just the newly imported one). If I select the object in the scene graph ui and select "Convert Root to Base Node(s)" the touch sensor starts working again.

I've tried reducing our rather complicated project into a minimal example to reproduce the issue, but I haven't succeeded yet, so maybe the code is run in some particular circumstance that triggers the issue, but I don't know what that circumstane might be

sigmunau, 2020-12-17:
[testcase.zip](https://github.com/cyberbotics/webots/files/5709396/testcase.zip)

I've uploaded a test case. worlds/testcase.wbt contains a world (this was generated by saving after running emulation with our real emulation, this causes some warnings in the webots console that doesn't appear in the proper simulaton)
minimal.py contains a minimal controller that moves a gripper down until the touch sensor triggers, then spawns a new object (far away from the touch sensor) and repeats. In my tests the touch sensor does not trigger on the second run.

sigmunau, 2020-12-17:
@lukicdarkoo Issues reproduced with R2021a December 15, 2020 installed from snap on ubuntu 20.04

oroulet, 2020-12-17:
I could reproduce with 2021a version.

omichel, 2021-01-20:
I have further simplified the provided [testcase.zip](https://github.com/cyberbotics/webots/files/5842042/testcase.zip) so that it doesn't display warnings any more and starts the controller automatically (easier for testing). I could indeed reproduce the bug.

stefaniapedrazzi, 2021-01-21:
In the `testcase` simulation, if I **reset the `WorldInfo.optimalThreadCount` to 1** then the bumper TouchSensor works correctly.
So it seems the issue could be in the code to update the spaces for the multi-threaded version of ODE. I will now investigate in this direction.

I also further simplified the `testcase` simulation, using only base nodes instead of PROTO, removing some intermediate nodes and unrelated objects, and we can now exclude that the issue is related to PROTO nodes.

sigmunau, 2021-01-22:
I  can confirm that the issue goes away if we set thread count to 1 in preferences also