cyberbotics/webots #2027: Mirroring Lidar output

SherlockSheep, 2020-07-29:
When I used webots with ROS, I got mirroring Lidar output when I viewed the scan topic in rviz, as can be seen at 6:12 in [video](https://www.youtube.com/watch?v=xRvFJDuhx3Y).

To make sure that the problem is not specific-sensor-related, I tried HokuyoUtm30lx and Sick LMS291,  and I met the same problem. I wonder why this happens and how I can fix it.

DavidMansolino, 2020-07-29:
You're right, the scan seems to be inversed, we will fix this for the next version of Webots that will be released soon.

DavidMansolino, 2020-07-29:
The problem is actually due to the fact that the Webots and RVIZ coordinate systems are not equivalent.
The Webots X-axis becomes -Y in RVIZ.
The Webots Y-axis becomes Z in RVIZ.
The Webots Z-axis becomes -X in RVIZ.

closed 2020-07-29 by DavidMansolino

SherlockSheep, 2020-07-29:
For velodyne VLP-16, I could tell :
> The Webots X-axis becomes -Y in RVIZ.
> The Webots Y-axis becomes Z in RVIZ.
> The Webots Z-axis becomes -X in RVIZ.

But for those single-line Lidars, the coordinate systems seemed to be weird.

SherlockSheep, 2020-07-29:
@DavidMansolino Thanks for replying. 
For velodyne VLP-16, I could tell :
> The Webots X-axis becomes -Y in RVIZ.
> The Webots Y-axis becomes Z in RVIZ.
> The Webots Z-axis becomes -X in RVIZ.

But for those single-line Lidars, the coordinate systems seemed to be weird.

DavidMansolino, 2020-07-29:
The coordinate system should be the same for the multi-lines and single-line lidars.
What difference do you see between them?

SherlockSheep, 2020-07-29:
Sorry my bad, I should post my screenshots.

This is how I placed Velodyne VLP-16:
[image]

This is how I placed HokuyoUtm30lx:
[image]

And I got these in rviz:
[image]

SherlockSheep, 2020-07-29:
In the third picture,:
z axis points toward the screen for the HokuyoUtm30lx rviz visualization.
z axis points outward the screen for the VLP-16 rviz visualization.