cyberbotics/webots #2033: qt_utils compilation fails on Ubuntu 20.04

stefaniapedrazzi, 2020-07-29:
**Describe the Bug**
Compiling qt_utils on Ubuntu 20.04 (debian and tarball) fails with this error:
```
gui/MainWindow.cpp:15:10: fatal error: X11/Xlib.h: No such file or directory
    15 | #include <X11/Xlib.h>
       |          ^~~~~~~~~~~
compilation terminated.
```

The package `libx11-dev` should be installed to make it work.
Not sure if it is really needed to include this library in the packages or we want to let users install it if they need it given that it seems to be used only in this library.

On Ubuntu 20.04 with the snap it works fine out of the box, as well as on Ubuntu 18.04.

omichel, 2020-07-29:
>  Not sure if it is really needed to include this library in the packages or we want to let users install it if they need it given that it seems to be used only in this library.

I would say that a regular user should not have to recompile this library. So, I would require this dependency only from a development environment. Therefore, I would not add it into the packages.

stefaniapedrazzi, 2020-07-29:
Ok, I agree.

Note that this test is still part of the Q&A, but since Qt robot windows are deprecated I would remove it from the Q&A too.

closed 2020-07-29 by stefaniapedrazzi