cyberbotics/webots #2037: Dll load error in pycharm

EnricoBeltramo, 2020-07-29:
**Describe the Bug**
Dll load error in pycharm

**Steps to Reproduce**
I create a project in pycharm starting from demo irb
I added the path as suggested in: https://cyberbotics.com/doc/guide/using-your-ide#pycharm:

my environment variables are: PYTHONUNBUFFERED=1;PATH=C:\Users\enrico.beltramo\AppData\Local\Programs\Webots\lib\controller\;C:\Users\enrico.beltramo\AppData\Local\Programs\Webots\msys64\mingw64\bin

(C:\Users\enrico.beltramo\AppData\Local\Programs\Webots\ is the installation path of webots)

I added C:\Users\enrico.beltramo\AppData\Local\Programs\Webots\lib\controller\python37 in content root

but I have anyway the follow error:
C:\Users\enrico.beltramo\.conda\envs\tf\python.exe "C:/Projects/Robottraiettorie/webox project/irb/controllers/inverse_kinematics/inverse_kinematics.py"
Traceback (most recent call last):
  File "C:/Projects/Robottraiettorie/webox project/irb/controllers/inverse_kinematics/inverse_kinematics.py", line 26, in <module>
    from controller import Supervisor
  File "C:\Users\enrico.beltramo\AppData\Local\Programs\Webots\lib\controller\python37\controller.py", line 15, in <module>
    import _controller
ImportError: DLL load failed

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: -

stefaniapedrazzi, 2020-07-30:
Are you running the Python controller in pycharm as extern controller?
Did you add `${WEBOTS}\lib\controller\python37` to the `PYTHONPATH` variable?

Here you can find all the needed steps to run extern controllers:
https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-language=python#environment-variables

And here is a link to similar questions posted on the Discord channel:
https://cyberbotics.com/doc/discord/technical-questions#geocsbi-07082020-11-17-10

EnricoBeltramo, 2020-07-30:
Yes, but I did with same result. If I run the python controller internally to webots, it works fine. Only if I try to use pycharm I have this problem. The controller package use only the mentioned dll folders? No others?
My version of python is 3.7 (conda), Webots R2020a revision 1 (never version can't be installed because for some reasons are blocked by the antivirus)

stefaniapedrazzi, 2020-07-30:
If you use Webots to run the controllers, then Webots sets automatically all the required environment values.
The point to make it work is to correctly set all the variables specified in the documentation:
* `WEBOTS_HOME`
* `PATH`
* `PYTHONPATH`

EnricoBeltramo, 2020-07-30:
I verified and all are set as suggested. I tried to uninstall and reinstall for all user and same error. At this point I don't know what to do again. 

Note: Anyway, the never version look not digitally signed: Win10 refuse to install it. Not this issue with previous version

DavidMansolino, 2020-07-31:
> Note: Anyway, the never version look not digitally signed: Win10 refuse to install it. Not this issue with previous version

You should still be able to install it by pressing on 'more information'.

stefaniapedrazzi, 2020-08-04:
@DavidMansolino fixed the documentation of the extern controller on Windows (#2050):
https://www.cyberbotics.com/doc/guide/running-extern-robot-controllers#environment-variables

It is also required to add `${WEBOTS_HOME}\msys64\mingw64\bin` and `${WEBOTS_HOME}\msys64\mingw64\bin\cpp` in the `PATH` variable.

Please reopen the issue if your problem should not be fixed.

closed 2020-08-04 by stefaniapedrazzi

EnricoBeltramo, 2020-08-05:
The folder ${WEBOTS_HOME}\msys64\mingw64\bin was already included, but I haven't the folder ${WEBOTS_HOME}\msys64\mingw64\bin\cpp in installation:

[image]

EnricoBeltramo, 2020-08-05:
> > Note: Anyway, the never version look not digitally signed: Win10 refuse to install it. Not this issue with previous version��
> 
> You should still be able to install it by pressing on 'more information'.

This option is available in a normal personal PC, but in factory's PC, for security reasons,  the installation of not-signed SW is blocked.

DavidMansolino, 2020-08-05:
Which version of Webots are you using?

EnricoBeltramo, 2020-08-05:
> Which version of Webots are you using?

Webots R2020a revision 1, my internal IT block installation of never versions due missing signed installation.

omichel, 2020-08-05:
It is strange because neither R2020a revision 1 nor R2020b are signed...