cyberbotics/webots #2047: Trouble with external controller and environmental variables in Windows with new Webots version R2020b

RobSol93, 2020-07-30:
**Describe the Bug**
After installing the new version R2020b (from R2020a), I get the following error when running my external controller:

> Traceback (most recent call last):
>   File "robot_controller.py", line 3, in <module>
>     from controller_base import WebotSupervisor
>   File "D:\Robert\Documents\Currence\Webots\WorldGit\emulation2.0\controllers\robot_controller\controller_base.py", line 4, in <module>
>     from controller import Supervisor
>   File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
>     import _controller
> ImportError: DLL load failed: ImportError: DLL load failed: The specified module could not be found.

I have [set the environmental variables](https://cyberbotics.com/doc/guide/running-extern-robot-controllers?tab-os=windows&tab-language=python#environment-variables), tried to create a new pycharm project, as well as running my controller from the terminal (outside pycharm).

I got the same error when I started using Webots (R2020a) a few months ago, but setting the variables fixed it.

**System**
 - Operating System: Windows 10

omichel, 2020-07-30:
In addition to `${WEBOTS_HOME}\lib\controller` and `${WEBOTS_HOME}\msys64\mingw64\bin`, can you try to add `${WEBOTS_HOME}\msys64\mingw64\bin\cpp` to your `PATH` environment variable and let us know if that fixes the problem?

DavidMansolino, 2020-07-30:
I tested and I had the same issue. Adding ``${WEBOTS_HOME}\msys64\mingw64\bin\cpp``to the Path fixes the issue. That's strange that we didn't saw this in the QA, I will fix the doc.

RobSol93, 2020-07-31:
Yes, worked for me as well. Thanks a lot!

DavidMansolino, 2020-07-31:
Thank you for the feedback!

PR #2050 (2020-07-31): Fix Windows Extern Controller Doc
Fix #2047
closed 2020-07-31 by DavidMansolino