cyberbotics/webots #2048: import 3d error, windows system crashed and cannot repair

wangzhankun, 2020-07-31:
**Describe the Bug**
Yesterday I run webots and import from 3d model (obj format) and windows crashed. Firstly it prompts that an error code (I forget what it is) and my windows resolution has been changed. I have to reboot my windows. And today I find there are many errors in my windows. My edge based on chrome cannot be used to search in the search bar for that if I try to search in the search bar it will crash.  My mission board setting is also changed. There may be other errors in my windows system but I don't find till now. Also there are some other issues that I cannot describe exactly in English. I think the webots may damage my windows settings and I don't know what to do.

**Steps to Reproduce**
1. import 3d model with obj format
2. waiting 
3. crashed

**System**
 - windows 10 1909
 - NVIDIA GeForce 940mx
- i7 7th Gen
- 12 GB RAM
- obj model is 362MB
- webots 2020b

DavidMansolino, 2020-07-31:
Can you please share the obj file you tried to import?
If you want to keep it confidential you can send it directly to support@cyberbotics.com

wangzhankun, 2020-07-31:
I send the file. Please check it.

DavidMansolino, 2020-07-31:
Still not received..

wangzhankun, 2020-07-31:
Resent it. The gmail didn't receive the mail due to the attach file is too large and I upload it into 163 mail cloud. You may have the download URL.

DavidMansolino, 2020-07-31:
Thank you, I confirm we received it now.

DavidMansolino, 2020-07-31:
I check your file, and it seems it is quite big, it does contain several million triangles, which is too much for Webots.
I would recommend reducing the resolution of the biggest meshes, in particular, those ones:
```
WARNING: mesh 'RM2020-2_Asm_RM2020-2_Asm_Dark_Gray' has more than 100'000 vertices, it is recommended reducing the number of vertices.
WARNING: mesh 'RM2020-2_Buff_Red_Blue_Rm2020-2_Asm.002_Dark_Gray.001' has more than 100'000 vertices, it is recommended reducing the number of vertices.
WARNING: mesh '2020_Infantry_2020_Infantry_Big_7' has more than 100'000 vertices, it is recommended reducing the number of vertices.
WARNING: mesh 'SentryFull_SentryFull_Big_7' has more than 100'000 vertices, it is recommended reducing the number of vertices.
WARNING: mesh '2020_Infantry.001_2020_Infantry.001_Big_7.001' has more than 100'000 vertices, it is recommended reducing the number of vertices.
```

closed 2020-08-03 by DavidMansolino