cyberbotics/webots #2057: Python interpreter does not runs between controller steps

oroulet, 2020-08-02:
**Describe the Bug**
If starting  a thread in a python controller, the thread is not allowed to run between controller steps. This is not the expected behaviour of a python script and breaks communication with external sytems. For example if the thread is supposed to receive UDP messages from another system, the UDP messages are not going to be received when expected

Most probably there is some C/C++ code sleeping with the Python lock. This must be fixed

**Steps to Reproduce**

Start webots and run the following script. The Thread is supposed to increment a counter 100 times a second, but in reality it does not even run once every second controller step (This is in fact even stranger, I was expected it to run a little at every step))

```
import time
from threading import Thread

from controller import Supervisor

class T(Thread):
    def __init__(self):
        Thread.__init__(self)
        self.start_time = None
        self.counter = 0
        self.ev_stop = False

    def run(self):
        self.start_time = time.time()
        while not self.ev_stop:
            time.sleep(0.01)
            self.counter += 1
            print(time.time())

    def __enter__(self):
        self.start()

    def __exit__(self, *args):
        self.ev_stop = True
        self.join()
        total = time.time() - self.start_time
        print(f"\n\n time elapsed: {total}s, nb counts: {self.counter}\n, counter per seconds: {self.counter / total}\n\n")

s = Supervisor()

with T() as t:
    while s.step(500) != -1:
        pass

```

**Expected behavior**

The Python thread should run freely independently of the controller period

**Additional context**
This seems to be related to https://github.com/cyberbotics/webots/issues/1903
With a very long controller period it becomes easy to reproduce it. Probably when the ctrl-C happens while the C code has the python lock or something similar

omichel, 2020-08-03:
Beware, the print command is sending output to the Webots console through the main thread. Thus, the problem may come from there. Can you try to execute the same program but instead of using print, try to print data into a file or send data to a socket?

oroulet, 2020-08-04:
@omichel do I need to do something so complicated? I just removed the print and the result is exactly the same:

python ex.py 
^C

 time elapsed: 10.886824607849121s, nb counts: 12
, counter per seconds: 1.1022497773453894

The thread could only increment counter 12 times in 10 s instead of the expected 1000 times.

omichel, 2020-08-05:
Yes, no need to further investigate about the print command, it's not the culprit. Maybe the [mutexes](https://github.com/cyberbotics/webots/blob/master/src/lib/Controller/api/robot.c#L639-L675) of the libController could be the cause of this problem... Could you try to disable them to see if that fixes the problem? You will need to modify this [robot.c](https://github.com/cyberbotics/webots/blob/master/src/lib/Controller/api/robot.c) file and recompile the libController from the [source](https://github.com/cyberbotics/webots/wiki).

lukicdarkoo, 2020-11-30:
@oroulet let us know if you have solved the problem. I am closing the issue, but feel free to reopen it.

oroulet, 2020-11-30:
I have not had time to look at this, we are using multiprocessing to work around that issue. If you did not change your code, you definitely still have the issue.

lukicdarkoo, 2020-11-30:
In that case, I will reopen it, so we can investigate it later

omichel, 2021-05-26:
I found a workaround allowing the thread to execute almost properly:
```python
with T() as t:
    while s.step(32) != -1:
        sleep(0.01)
        pass
```
However, I still don't understand why the step function is taking all the CPU time, I will continue to investigate. Meanwhile, could you tell @oroulet if this solution is workable for you?

omichel, 2021-05-26:
After some research, it seems to be a standard problem of multi-threading. If the main thread is busy, which is the case, it should make explicit calls to [sched_yield()](https://man7.org/linux/man-pages/man2/sched_yield.2.html) to yield time to other threads. In Python this can done with time.sleep(0). It should be possible to add this into the libController, so that you don't have to do it in Python or C controllers.

oroulet, 2021-05-26:
Great that seems to be a much easier issue than what I was afraid of. We will test. Thanks

oroulet, 2021-05-27:
@omichel OK I had a look at it.
I do not think the proposed solution works. It might have looked OK in your example with a 32ms step but if you run with a 500ms step you will see that the python thread does not run while step() is excuting: The code above is supposed to return without a counter per second which is close to 100. Not 1.9 ;-)

```

 time elapsed: 6.218572616577148s, nb counts: 12
, counter per seconds: 1.9297032839997754

```
Our issue (And probably others) is that we need a python thread that is reactive even while step() is running. It looks like your Supervisor.step() take the python global lock during its entire execution()

omichel, 2021-05-27:
Yes, this is because performing a simulation step of 500 ms in a single shot is not too large. Instead I would recommend you to split it into basic time steps:
```python
time_step = s.getBasicTimeStep()
with T() as t:
    for i in range(500 / time_step):
        if s.step(time_step) == -1:
            break
        time.sleep(0)  # allow Python to yield time to other threads
```

oroulet, 2021-05-27:
500ms was just to highlight the issue. I found out I could get the example over (and our code btw!!!) to behave almost like expected with something like

```
    while s.step(64) != -1:
        time.sleep(0.064)
```

So it looks like your Supervisor.step() method keep the global interpreter lock while sleeping in C++  up to the int argument of step(). So if we sleep in python for almost the same time, we manage to avoid webots locking everything while sleeping.

Seems to work, but I would rather get your opinion on what is happening here. if we sleep for 0.064s and your step() method also take some time, we might not be calling the step() at the expected frequency expected by webots()

omichel, 2021-05-28:
It seems @sigmunau found a very good solution to this issue.

oroulet, 2021-05-28:
if you are fine with this, then it is good. It is not tested but in theory it should work. Btw Sigmund is my colleague

sigmunau, 2021-05-28:
> if you are fine with this, then it is good. It is not tested but in theory it should work. Btw Sigmund is my colleague

I've tested it with your testcase from the issue description
```
1622185019.5992935
1622185019.6095414
1622185019.6197023
1622185019.6298811
1622185019.6401088
1622185019.6503932
1622185019.660652
1622185019.6708608
1622185019.6811738
1622185019.6914716
1622185019.701739
```

oroulet, 2021-05-28:
I meant we have not tested it with our robot emulation, but in theory in it should work

omichel, 2021-06-07:
Fixed in #3104.

PR #3104 (2021-05-31): Release python global interpreter lock when running c++ code
Replaces #3103 aiming at the develop branch instead of the master branch.

> Fixes #2057. Comes at the cost of potentially increased overhead. See http://swig.10945.n7.nabble.com/How-to-release-Python-GIL-tp5027p5029.html
closed 2021-06-07 by omichel