cyberbotics/webots #2069: Procedural PROTOs not Working on Windows Release

DavidMansolino, 2020-08-04:
**Describe the Bug**
Some of the procedural PROTO are not working with the latest release of Webots for Windows.

For example in the ``humanoid_marathon``, the following warning is raised:
```
FILE_NAME.proto': error: Template engine error: luaL_dostring error : ...m Files/Webots-2020-b/resources/lua/liluat/liluat.lua:520: attempt to perform arithmetic on local 'line_number' (a nil value).
```

This is reproducible with the sample PROTO of the documentation too: https://www.cyberbotics.com/doc/reference/procedural-proto-nodes#example

omichel, 2020-08-04:
Fixed in #2070.

jnoyola, 2021-08-28:
I'm seeing this issue again when using lua gd in Windows R2021b.