cyberbotics/webots #2080: Ctr-C on a external robotcontroller with a camera kills the camera

oroulet, 2020-08-05:
**Describe the Bug**
Ctr-C on a robotcontroller with a camera kills the camera. There is no way to make the camera work again without reloading the mode. Probably related to https://github.com/cyberbotics/webots/issues/1903 but this time a simple ctrl-c trigger the issue.

There is a bypass so mainly reporting so you check this is not hiding a dangerous bug

**Steps to Reproduce**
open attached simple model with one camera and a cube
run following simple code

```
from controller import Supervisor
from webot_cameras import WebotCam3d

Supervisor = Supervisor()
time_step = 10

robot1 = Supervisor.getFromDef("MyRobot")
robot1.gripper_3d_cam = WebotCam3d("gripper_3d_cam")

Supervisor.simulationSetMode(Supervisor.SIMULATION_MODE_REAL_TIME)

def simple_run_simulation():
    Supervisor.getKeyboard().enable(time_step)
    Supervisor.simulationSetMode(Supervisor.SIMULATION_MODE_REAL_TIME)

    while Supervisor.step(time_step) != -1:
        pass

try:
    simple_run_simulation()
finally:
    Supervisor.simulationReset()
```
double-click on  camera to show the camera controller in UI. enable camera, move the box to see it is working fine

ctrl-C on controller
the message "INFO: Terminating extern controller for robot "robot"." pops up so exit was clean.

restart controller

=> the camera is dead, no way to make it see something

if instead of ctrl-c we click on reset in UI then controller dies and we can restart it and camera works

**Expected behavior**

The camera should work again since exit was clean. There is obviously a cleanup bug somewhere.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

proto:
```
#VRML_SIM R2020a utf8
# license:
# license url:
# tags: 
# RealSense d415

PROTO RSd415 [
    field SFString      Name_3D_Cam         "gripper_3d_cam"
    field SFVec3f       Translation_3D_Cam  0 0 0
    field SFRotation    Rotation_3D_Cam     0 1 0 1.5708
    field SFInt32       Width_3D_Cam        1280
    field SFInt32       Height_3D_Cam       720
    field SFFloat       FOV_3D_Cam          1.13446
    field SFFloat       minRange_3D_Cam     0.16
    field SFFloat       maxRange_3D_Cam     10

    field SFString      Name_2D_Cam         "gripper_2d_cam"
    field SFVec3f       Translation_2D_Cam  0 0 0
    field SFRotation    Rotation_2D_Cam     0 1 0 1.5708
    field SFInt32       Width_2D_Cam        1280
    field SFInt32       Height_2D_Cam       720
    field SFFloat       FOV_2D_Cam          1.21126
]
{

Transform {
    translation -0.07 0.02 0
    rotation    0 0 1 0.366519143
    
    children [
        DEF cam_3d RangeFinder {
            name IS Name_3D_Cam
            translation IS Translation_3D_Cam
            rotation IS Rotation_3D_Cam
            
            width IS Width_3D_Cam
            height IS Height_3D_Cam

            fieldOfView IS FOV_3D_Cam

            minRange IS minRange_3D_Cam
            maxRange IS maxRange_3D_Cam
        }

        DEF cam_2d Camera {
            name IS Name_2D_Cam
            translation IS Translation_2D_Cam
            rotation IS Rotation_2D_Cam
            
            width IS Width_2D_Cam
            height IS Height_2D_Cam

            fieldOfView IS FOV_2D_Cam
        }

        RSd415_shape {
            
        }        

    ]
}

}
```

world:

```
#VRML_SIM R2020b utf8
WorldInfo {
  basicTimeStep 10
  optimalThreadCount 2
  coordinateSystem "NUE"
}
Viewpoint {
  orientation -0.845084681024824 0.10710956116385281 -0.5237933025578488 0.1196038334325319
  position 0.10404933575989003 0.19288589190769062 2.6282182169497323
}
TexturedBackground {
}
Solid {
  translation -0.43 0 0
  children [
    Shape {
      appearance Appearance {
        material Material {
        }
        texture ImageTexture {
          url [
            "C:/Users/Robert/Desktop/Uten navn.png"
          ]
        }
      }
      geometry Box {
        size 0.1 0.1 0.1
      }
    }
  ]
}
DEF MyRobot Robot {
  translation 0.42 0.25 0
  children [
    RSd415 {
    }
    Shape {
      appearance Appearance {
      }
      geometry Box {
        size 0.1 0.1 0.1
      }
    }
  ]
  controller "<extern>"
  supervisor TRUE
}
TexturedBackgroundLight {
}
```

DavidMansolino, 2020-08-06:
Which version of Webots are you using?
We fixed a bug which looks very similar to what you are describing in R2020b (#1802).

oroulet, 2020-08-06:
I tested both current stable and 2021 nightly. Same behavior

DavidMansolino, 2020-08-07:
The problem is actually not the camera itself, if you look at the camera overlay you can see that it is still updated at each step.
The problem is the robot-window that is somehow still connected to the old controller and therefore doesn't receive any update.

stefaniapedrazzi, 2020-08-20:
Could you please try again using the latest nightly build?
https://github.com/cyberbotics/webots/releases/tag/nightly_19_8_2020

I recently fixed some issues with initialization of external camera windows.
I tried to reproduce this issue, but it seems to work now: even after restarting the controller the image in the camera window is updated correctly.

oroulet, 2020-08-20:
We will try asap

DavidMansolino, 2020-08-20:
I am not sure I am not sure  #2119 will help as this issue is about extern controller and not external window.

stefaniapedrazzi, 2020-08-20:
Based on the description is about extern camera window not working after killing extern controller. it is good o to give it another try.

DavidMansolino, 2020-08-20:
> Based on the description is about extern camera window not working after killing extern controller. it is good to give it another try.

Oh yes you're right, since the problem was also present in the robot window I thought it was not related.

oroulet, 2020-08-20:
Ok. Need to wait to be back with good network. Will test next week

oroulet, 2020-08-25:
@stefaniapedrazzi No change wiht latest nightly. Ctrl-C on the controller, although getting the message "INFO: '<extern>' controller exited successfully." the cameras are dead

DavidMansolino, 2020-08-28:
@oroulet, this has now been fixed in #2152, a patched version will be available in the nightly builds from tomorrow: https://github.com/cyberbotics/webots/releases

closed 2020-08-28 by DavidMansolino