cyberbotics/webots #2085: Segmentation error when changing the type of the floor

agayev169, 2020-08-06:
I was following the official [tutorials](https://cyberbotics.com/doc/guide/tutorial-2-modification-of-the-environment) and after creating a floor, I tried to change its type. When changing the type of the floor to any other value different from the default (chequered) I always get the following error:
```/usr/local/bin/webots: line 88:  5916 Segmentation fault      (core dumped) "$webots_home/bin/webots-bin" "$@"```

My OS is Ubuntu 18.04, no graphics card.

stefaniapedrazzi, 2020-08-06:
Which version of Webots are you using?

We fixed many related issues in the latest Webots R2020b.
If you are using a previous version, could you please check if you get the same errors on Webots R2020b?

agayev169, 2020-08-06:
I am using Webots R2020b.

stefaniapedrazzi, 2020-08-06:
I will properly fix the issue as soon as possible.

But probably the changes in #2084 should already prevent this issue to arise in case of the Floor PROTO.

agayev169, 2020-08-06:
I read PROTO files both for Floor and RectangleArena. As far as I understood, the variable ```floorTextureUrl```, whose problem is described in #2084 is the same as ```texture``` in Floor PROTO file. However, the same code pattern is used in the Floor.PROTO file [here](https://github.com/cyberbotics/webots/blob/323d9231584325b0da9106c2bea2f91a90199df7/projects/objects/floors/protos/Floor.proto#L29-L32).

agayev169, 2020-08-06:
FYI: I tried to modify the world file where I changed the type value of a Floor object and it worked. I thought it may help during debugging.

stefaniapedrazzi, 2020-08-06:
Yes, it should be sufficient to remove the deprecated fields warning in the PROTO file to let it work.

This will workaround the issue in this particular case. I found the source of the issue but may take some time to correctly fix it and test it.

PR #2100 (2020-08-10): Fix use of invalid scene tree selected item
Fix #2085: fix missing check for invalid selected tree item.
closed 2020-08-10 by stefaniapedrazzi