cyberbotics/webots #2090: The problem of buliding omni wheel robot

huiduan-25, 2020-08-06:
I try to build an omnidirectional robot  with 6 omni wheels. After assembly, there is only one rotational motor shown on the graph, I have no idea what part is wrong. Could you please help me?

Thank you

huiduan-25, 2020-08-06:
[image]

DavidMansolino, 2020-08-07:
Did you set different names for each motor or did you leave the default one?

huiduan-25, 2020-08-07:
> Did you set different names for each motor or did you leave the default one?

Hi，David
Each motor has the individual name from Wheel1 to Wheel6. But also，I am confused about why the name on the graph is rotational motor 0.

DavidMansolino, 2020-08-07:
Can you share the world file here?

huiduan-25, 2020-08-07:
Dear Mr. David

Hope this email finds you well.

I have attached the world file here. I appreciate your kind help.

Best wishes

huiduan-25

huiduan-25, 2020-08-07:
[try 9.zip](https://github.com/cyberbotics/webots/files/5040276/try.9.zip)
Hi, I am not sure if the file from the outlook is available. I attach the file here as well.

Thank you

huiduan-25, 2020-08-07:
Dear Mr. David

I just attach the zip file in the email.

best wishes

hudiaun-25

DavidMansolino, 2020-08-07:
You set the DEF name but not the actual name, you have to define the ``name`` field of the ``RotationalMotor`` nodes.

huiduan-25, 2020-08-07:
Dear Mr. David

That is really helpful！The problem has been solved！I can see full 6 graphs now. But I am not sure what the meaning of the 0 represents in the picture.
[cid:dc8a8a6c-e1cf-4977-950a-e244db92a275]
Once the world is run, the robot starts to the rotation movement, to make the robot move forward, should I apply the controller to the robot.  Is that possible I can refer the controller code of omniwheel from Github.

Appreciate again for your help.

Best wishes

huiduan-25

获取 Outlook for iOS<https://aka.ms/o0ukef>

DavidMansolino, 2020-08-07:
0 is the current target position.

About controlling the robot, I would strongly recommend following the tutorials, in particular this one: https://www.cyberbotics.com/doc/guide/tutorial-4-more-about-controllers

closed 2020-08-07 by DavidMansolino

huiduan-25, 2020-08-07:
That is great! Thank you for your help!

huiduan-25