cyberbotics/webots #2091: Python controller file not located

lordNil, 2020-08-07:
Hello 

please see the attached image for the console output. 

This is a new install of webots 2020b
The program is not able to open controller.py
I have verified that controller.py exists in the specified path. 

`Traceback (most recent call last):
  File "my_controller.py", line 3, in <module>
    from controller import Robot
  File "C:\Program Files\Webots\lib\controller\python37\controller.py", line 31, in <module>
    import _controller
ImportError: DLL load failed: The file cannot be accessed by the system.
WARNING: 'my_controller' controller exited with status: 1.`

DavidMansolino, 2020-08-07:
Are you starting the controller the usual way (webots starts it) or are you using an extern controler?

lordNil, 2020-08-07:
the controller is the usual way within webots. 

I fixed this problem my going back to a windows restore point. 
And then clean install of python (Anaconda) + webots fixed it. 

I had uninstalled and re-installed webots  (and python) before, and I guess the previous version was messing with the current version. The program would be buggy and could not run any controller reliably.

omichel, 2020-08-07:
OK, thank you for the feedback. Please note that Webots only supports the 64-bit version of the official Python (from https://python.org) out of the box. For other variants, including Anaconda, it may or may not work. See the documentation for details on how to [use an alternative python version](https://www.cyberbotics.com/doc/guide/using-python#use-an-alternative-python-version). Did you apply this method to make it work with Anaconda or did it worked out of the box?

closed 2020-08-07 by omichel

lordNil, 2020-08-07:
I did not mess around with SWIG for Anaconda. 

Anaconda (the latest one with python 3.8) seemed to work out of the box. 
During install, I included the option to add Anaconda to windows PATH.

omichel, 2020-08-07:
OK, very good. Thank you for the feedback.