cyberbotics/webots #2093: MJPEG streaming not working with local streaming viewer

omichel, 2020-08-07:
When starting Webots in stream mode and connecting with the local streaming viewer, the X3D mode works fine, but the MJPEG one doesn't:
```
ERROR: Streaming server: Unsupported message: mjpeg: 1215x728.
```

stefaniapedrazzi, 2020-08-10:
Do you mean using the `resources/web/streaming_viewer/index.html`?

It works fine for me on Linux in a branch based to a prior master branch.
So it the issue be introduced in #2077?

omichel, 2020-08-10:
> Do you mean using the resources/web/streaming_viewer/index.html?

Yes.

> It works fine for me on Linux in a branch based to a prior master branch.
> Has the issue been introduced in #2077?

OK, thank you for the test. I will check.

omichel, 2020-08-10:
It works actually on the master branch as well.
The reported error message appears only if you forget to specify `mode=mjpeg` when starting Webots with the `--stream` option, which is fine. I will maybe try to improve it a little bit, so that it's clear how to fix it.

PR #2105 (2020-08-10): Improve streaming error messages
Fixes #2093.
This should help the user to detect a mode mismatch between Webots and the web client.
closed 2020-08-10 by omichel